arkypita/LaserGRBL #1442: Can't Home since no Z Axis in laser machine

BalmierLight, 2021-04-30:
Hi, so I recently bought a laser machine, not sure of the company/model tbh, but most likely TWOTREES TOTEM 5.5w (maybe 300mmx300mm)
So when I do homing, the x axis moves a bit, then stops, and says:
"Homing Fail. Could not find limit switch within search distances.
Try increasing max travel or decreasing pull-off distance or check wiring."
it doesn't even try to move the y axis (also I'm not sure if my engraver has Limit Switches)
For the configs, here it is:
`$0=10 (Step pulse time)
$1=25 (Step idle delay)
$2=1 (Step pulse invert)
$3=2 (Step direction invert)
$4=0 (Invert step enable pin)
$5=0 (Invert limit pins)
$6=0 (Invert probe pin)
$10=1 (Status report options)
$11=0.010 (Junction deviation)
$12=0.002 (Arc tolerance)
$13=0 (Report in inches)
$20=0 (Soft limits enable)
$21=1 (Hard limits enable)
$22=1 (Homing cycle enable)
$23=0 (Homing direction invert)
$24=25.000 (Homing locate feed rate)
$25=500.000 (Homing search seek rate)
$26=250 (Homing switch debounce delay)
$27=1.000 (Homing switch pull-off distance)
$30=1000 (Maximum spindle speed)
$31=0 (Minimum spindle speed)
$32=1 (Laser-mode enable)
$100=80.000 (X-axis travel resolution)
$101=80.000 (Y-axis travel resolution)
$102=250.000 (Z-axis travel resolution)
$110=12000.000 (X-axis maximum rate)
$111=12000.000 (Y-axis maximum rate)
$112=2000.000 (Z-axis maximum rate)
$120=6000.000 (X-axis acceleration)
$121=6000.000 (Y-axis acceleration)
$122=10.000 (Z-axis acceleration)
$130=300.000 (X-axis maximum travel)
$131=300.000 (Y-axis maximum travel)
$132=200.000 (Z-axis maximum travel)
`
So yeah, any help appreciated, Thanks in advance!

gmmanonymus111, 2021-04-30:
If you are right about your engraver model, then it has no limit switches. Without limit switches, you can't home.
With limit switches, you need to configure your homing cycle (of is not configured already) by editing GRBL code or by choosing a pre-compiled firmware from LaserGRBL. In both cases, you are risking to mess up the current pinout of your machine (since TWOTREES TOTEM 5.5w uses a custom nano board I think). This is a question the manufacturer could help you more.

BalmierLight, 2021-04-30:
@gmmanonymus111 Oh I see, I guess that explains why it just tries to keep moving when it hits the end of it. Aight, so in my current case, what is a possible option to do? (Oh, and when I installed the manufacturer Configs that came with the printer, it specified $130 and $131 as 170mm, although my laser is larger than that, also by it choosing 170mm, the sizes where inaccurate while printing, so if I'd specify a size of 230mm, it'd just go to like 180mm, so I manually set it to 300mm for both x and y, is that okay?
And tysm for the help!

gmmanonymus111, 2021-04-30:
Well, you could try install limit switches, if your control board has an input for them, and if you really need those.
The engraving area should not make the engraving inaccurate, and can be set as you see fit. Inaccurate engraving may be caused by wrong steps/mm values.

BalmierLight, 2021-04-30:
I see, from my board, It does not list Arduino anywhere tho, It's a custom-made one.

For the engraving size, I still don't get it, like when it was configured in the configs, the engraver was set ((X-Y-axis maximum travel)) as 170mm, and as I told you, it won't give me the right size I order, e.g. I sent a 200mm*200mm Pic, it'd just go all the way to 180mm at max, while when I reconfigured the engraver to ((X-Y-axis maximum travel)) as 300mm that would 98% solve the issue

gmmanonymus111, 2021-04-30:
Maybe it got resized based on your limits (AFAIK neither GRBL or LaserGRBL re-scale your image)? Strange, it should just stop when it reaches soft limits (if enabled) or do the job regardless of set maximum travel (if disabled)

closed 2021-06-21 by arkypita