arkypita/LaserGRBL #1445: Resize doesn t work

BeikircherDaniel, 2021-05-02:
The resizing rectangle starts outside the picture.

If I klick on the icon to resize an image , then I klick where I want to start to resize it but the rectangle is not opening there.

StuartB4, 2021-05-02:
[image]

BeikircherDaniel, 2021-05-02:
Thanks i know this but the box does not open where i want. It opens outsithe picture. On my pc with windows 7 was no problem now on my new one with windows 10 it does not work how it should.

StuartB4, 2021-05-02:
The only other way to resize an image is in Target image settings where you change the W and H settings to set the engraving / cutting size. If you change one of them the other will change so the image stays in the correct size ratio.

StuartB4, 2021-05-02:
I have windows 10 and it works ok.

BeikircherDaniel, 2021-05-02:
Yes this is clear but if i want to cut out the image in Lasergrbl this is not the solution. I try to make a screenshot to show what the problem is.

BeikircherDaniel, 2021-05-02:
[image]

BeikircherDaniel, 2021-05-02:
This is how it looks with the left mouse key holded down.

closed 2021-05-04 by BeikircherDaniel

ibisdave, 2023-12-13:
I am having the same issue how was this fixed?

BeikircherDaniel, 2023-12-13:
It wasnt fixed but it cuts where you want. Just the line previev is not where it should be.

ibisdave, 2023-12-13:
I have the software on my office computer and it works fine it is my shop computer that does exactly what you had said. It does work it just puts the box in the wrong spot