arkypita/LaserGRBL #1449: LaserGRBL does not do 1 pass in Laser cut

nissimzur, 2021-05-04:
I create a cut for plastic sheet. See attached. 
The line is 0.3mm. 
1 pixel on Photoshop that was convert to SVG.
From unknown cause the LaserGRBL do 2 or more cut passes.  
Why? This is just double and triple the cut time and the cut is wider because of the temperature neer cut line. 
Thank you
Dr. Nissim Zur
[0.3 mm-cut.zip](https://github.com/arkypita/LaserGRBL/files/6419790/0.3.mm-cut.zip)

gmmanonymus111, 2021-05-04:
I'm 99% certain that LaserGRBL is not at fault here. It is your method generating the SVG file, and it generates 2 lines very close to each other. Generating with InkScape, my SVG file is cut only once.
Side note: There is no need for line width setting when engraving SVG files. Line width equals to your laser point (or rectangle).
[tiger.zip](https://github.com/arkypita/LaserGRBL/files/6420475/tiger.zip)

nissimzur, 2021-05-09:
Maybe  LaserGRBL is not at fault. But every SVG line has two sides. And LaserGRBL tries to cut both sides. No matter how the line is tiny.

arkypita, 2021-05-09:
> ... But every SVG line has two sides...

This is not true. SVG lines are line and does not have sides.

closed 2021-06-21 by arkypita