arkypita/LaserGRBL #1450: LaserGRBL Ramdom stops after 6 minutes of working

consol3, 2021-05-04:
Anyone knows what this "error" could mean and what could cause it? Also how can I get rid of it?
The LaserEngraver just randomly  stops after 5-6 minutes of working and I have to reset it everytime.
[image]

gmmanonymus111, 2021-05-04:
With the info provided, my guess is electrical noise, or your controller is overloaded. 
What type of connection you use? (USB, telnet etc)
What is your controller board/engraver? (Arduino nano/uno/mega based build, or some pre-built machine)

arkypita, 2021-05-04:
This issue is widely described in lasergrbl FAQ: https://lasergrbl.com/faq/

Did you read them?

consol3, 2021-05-04:
> This issue is widely described in lasergrbl FAQ: https://lasergrbl.com/faq/
> 
> Did you read them?

I did...but I don't know which one you are talking about, it's my first time in GRBL and I don't really know how to use it but I did set up everything just right. If you're talking about the limit switches, I don't know how to disable it , I don't have limit switches.

consol3, 2021-05-04:
> With the info provided, my guess is electrical noise, or your controller is overloaded.
> What type of connection you use? (USB, telnet etc)
> What is your controller board/engraver? (Arduino nano/uno/mega based build, or some pre-built machine)

USB cable, it came prebuilt but it doesn't look like any arduino I've seen before tho...it's a 3018 CNC but the card seems different from the one I found online :(

gmmanonymus111, 2021-05-04:
Does it have an offline controller/screen? (Because some 3018's have them)
In case it has (quoted from FAQ):

> Protocol between LaserGRBL and the machine does not foresee that there are multiple systems trying to control the laser. If LaserGRBL is controlling the laser any other display or controller on the machine can only disturb the communication.
...
> Machines equipped with an offline control display require this to be turned off or disabled in order to use them with the LaserGRBL.

arkypita, 2021-05-04:
> I did...but I don't know which one you are talking about, it's my first time in GRBL and I don't really know how to use it but I did set up everything just right. If you're talking about the limit [...]

Charapter "Problem detected message: hang or disconnect during engraving"

consol3, 2021-05-04:
> Charapter "Problem detected message: hang or disconnect during engraving"

Yup, thank you, I did read it out slowly right now and did everything it said...still...I does stop :( The problem is that I have a CH340 instead of a Legit Arduino board....so that's probably the big problem...as nothing else works...guess it's time for a refund :((

consol3, 2021-05-04:
> Does it have an offline controller/screen? (Because some 3018's have them)
> In case it has (quoted from FAQ):

Nope, doesn't have one. I did read the FAQ like someone else suggested...and I found out that I have a CH340 board and not a legit Arduino one...guess the Chinese stuff will always be Chinese stuff haha...I'll try a return or change the board :( Thank you for your answers tho <3

gmmanonymus111, 2021-05-04:
CH340 is only a USB to serial converter. Arduino uses ATmega chips (ATmega328 for uno iirc), and some comes with the same CH340 USB to serial converter (like uno+WiFi R3, or some esp boards). The exact type is written on the chip. Could you and a picture of the control board?
In the meantime, you could try with a different USB cable, if you haven't tried already.

arkypita, 2021-05-04:
Not all board with CH340 have these problems, I have several that work fine.

Before sending it back try to verify that the problem is not due to the cable or some interference (as it is 90% of the time)

consol3, 2021-05-05:
> CH340 is only a USB to serial converter. Arduino uses ATmega chips (ATmega328 for uno iirc), and some comes with the same CH340 USB to serial converter (like uno+WiFi R3, or some esp boards). The [...]

Here is the Board, it does have a ATmega328P chip. Also I didn't try another cable cause I don't have another one, but the one I got looks exactly like the ones GRBL suggests, even with a ferrite core.

[image]
[image]

consol3, 2021-05-05:
> Not all board with CH340 have these problems, I have several that work fine.
> 
> Before sending it back try to verify that the problem is not due to the cable or some interference (as it is 90% of the time)

I don't know what else to try, I've tried setting it on Slow and RepeaOnError but still stopps, not as often though. I can't try another cable cause I don't have another one, could it be the fact that my laptop is right next to it?

gmmanonymus111, 2021-05-05:
Could you try runing it while the stepper motors are disconnected? (Or use `$C` to switch to check mode, which will process all commands, but moves nothing on your machine)
If it works that way (as the USB cable transmits data correctly), my second guess would be the PSU. (When steppers + laser is running, there is not enough power to keep everything working, and the control board resets;in the process, it generates unreadable data, too, at least similar happens on ESPs)

StuartB4, 2021-05-05:
Looks like a copy of as Woodpecker board.

[image]

It should work with the CH340 driver because it has the correct chip on the board.
The one just above the reset button. 

Are your laser connection wires in the correct order on the main board.
I cant see the wire colours or the writing on the board where the laser is plugged in.

consol3, 2021-05-05:
> Looks like a copy of a Woodpecker board.
> 
> [image]
> 
> It should work with the CH340 driver because it has the correct chip on the board.
> The one just above the reset button.
> 
> Are your laser connection wires in the correct order on the main board.
> I cant see the wire colours or the writing on the board where the laser is plugged in.

Looks just like it, also, I think the laser wiring is good as I can do photos just fine with Dynamic Mode. 
The colors are like this: 
[image]

consol3, 2021-05-05:
> Could you try runing it while the stepper motors are disconnected? (Or use `$C` to switch to check mode, which will process all commands, but moves nothing on your machine)
> ~If it works that way (as the USB cable transmits data correctly), my second guess would be the PSU. (When steppers + laser is running, there is not enough power to keep everything working, and the control board resets;in the process, it generates unreadable data, too, at least similar happens on ESPs)~
> EDIT: Nope, it appears your board does not reset, just shows random data.

I used $C and sometimes works, sometimes it doesn't...it's kinda weird tbh...also, I moved the laser away  from my computer and it stops once in like 20 minutes, which is way better than before...but still does it and that means I can't let it burn and go away :(

gmmanonymus111, 2021-05-05:
Well then either the cable, the USB port of your computer, or your control board is faulty, since the problem persists without inference from other electrical components. Change the cable, try another PC if possible before sending it back

nistre, 2021-06-11:
I Feel like this may be more of a systemic issue.  @consol3, what mode were you trying to engrave in? (i.e. vectorize, line-by-line, etc.)  I am having the exact same issue that occurs almost exactly at the 6-minute mark , plus/minus a millisecond: 
[image]

I never ran into this issue until trying to engrave with Line-by-line tracing.  With my exact same setup, I can do the same image in vectorize no problem.

closed 2021-06-21 by arkypita