arkypita/LaserGRBL #1455: SVG Import does not respect layer transformations

alastaira, 2021-05-10:
In LaserGRBL v4.3.0, when importing an SVG layer from Inkscape containing one or more layers in which transformations have been applied to the layer, it does not seem to respect those transformations.

Simple reproduction attached in zip file (archive only contains a single SVG file, but Github won't let you upload a file with SVG extension - doh!), in which the grid lines are placed in a separate layer ("Layer 1"), with translate(0.25,60.25) attribute. Text and outline are not contained within that layer. The SVG renders correctly in Inkscape and when loaded in a browser such as Microsoft Edge, but incorrectly in LaserGRBL as shown in screenshots.

[LaserGRBL-LayerBug.zip](https://github.com/arkypita/LaserGRBL/files/6451815/LaserGRBL-LayerBug.zip)

[image]
[image]
[image]

alastaira, 2021-05-10:
Possibly relevant - this SVG file was originally exported from OpenSCAD before being brought into Inkscape. And, even though it renders correctly within Inkscape, it therefore seems that some of its structure might differ from the way Inkscape would have defined the same file internally.

I notice that, when creating SVG files from scratch within Inkscape, all elements are placed within a layer (although I couldn't see this being a requirement according to the SVG standard).
However, in this file, I'm beginning to thing that the problem is actually with elements *not* inside a layer, but as direct child elements of the root <svg> tag.

StuartB4, 2021-05-10:
It loads properly if you group it all together in inkscape first. 

[image]

alastaira, 2021-05-11:
The more I look at this, the more I'm convinced that the problem arises as a result of the fact that the file originated from OpenSCAD, and its definition of the SVG "non-standard standard" differs from that which Inkscape uses (which I agree, should be the SVG version on which LaserGRBL is based).

Having identified the problem, there's a relatively simple workaround which is just to edit->cut all the objects and paste them into a layer in Inkscape (eitherthe default "Layer 1", or a new one). Then save the SVG again.

arkypita, 2021-05-11:
Hi. Currently the code that transforms the SVG into GCode is a code that I did not write myself, but that I copied "as is" from another project. My bugs are his bugs, and even if I try to fix some of them, since it is third-party code it is always very difficult to be able to intervene without risking to break something.

If we also assume that this code is not written very well, and the difficulty of interpreting the SVG specification, it turns out that I prefer to intervene as little as possible on this part.

I think the "final" step would be to throw this code away and replace it with SVG reading done with a better and more supported library [like this one](https://github.com/svg-net/SVG) and implement my own "from svg to gcode" routines.

closed 2021-05-11 by alastaira

svenhb, 2021-11-21:
>My bugs are his bugs

This code (GCodeFromSVG.cs) was taken in 2018 and has improved since them. Actual result of this excample looks like this:
[image]

falias4, 2021-11-21:
### Regarding this specific issue
The `transform="translate(x,y)"` property isn't interpreted in this version (from 2018) of the `GCodeFromSVG.cs`.
So a quick fix would be to add specific code to handle this.

### General update of the 'external library'
Out of curiosity I tried to integrate the latest code from [GRBL-Plotter](https://github.com/svenhb/GRBL-Plotter) but it was already pretty exhausting to get a simple result. 🙈
The whole thing got greatly reworked and there are many new files and references to classes, settings, etc. which are very "deep" inside GRBL-Plotter.

**Don't want to give false hope: I probably won't work on this any further.**
I don't have enough knowledge about GRBL-Plotter and all the settings to do that in a reasonable amount of time.... although the result would be very cool.
PS: The hatch fill is activated by accident. No idea how to turn it off. 😅
[image]

svenhb, 2021-11-21:
>PS: The hatch fill is activated by accident. No idea how to turn it off.

Can be deactivated here, for more options check https://grbl-plotter.de/index.php?id=form-setup&setlang=en
[image]

https://grbl-plotter.de/index.php?id=form-setup&setlang=en

arkypita, 2021-11-30:
Thank you for this report and thanks to @svenhb for the code, and thanks @falias4 for the fix.
I am currently very busy but as soon as I can I do a test and compile a new version.

PR #1658 (2021-12-08): Fix wrong SVG positions due to transform attribute
TL;DR: I simply replaced the functions regarding the transform-attribute with the latest ones from GRBL-Plotter.

---
As discussed in #1455 the `GCodeFromSVG.cs` is quite old and a whole rework would take much time.
So I specifically searched for the reported issue and simply replaced the functions with the new ones.

Basically only had to change the `ParseTransform` so that it returns the `Matrix` instead of a `bool`.
Other small changes are due to changes of function names (casing) and that they are static now.

Should fix #1455 and #1651