arkypita/LaserGRBL #1458: Keywords [left] [right] etc. do not work with inserted gcode file.

branther, 2021-05-16:
i use LasrGRBL with version 4.3.0
When I create a gcode file from Inkscape with Jtech Laser Tool, the dimensions are not taken over.
I cannot use the [left], [right] ...etc. functions.
When I vectorize a jpeg file with LaserGRBL the macros work correctly.
Is this ok or am I doing something wrong?
a simple rectangle e.g. created with the extension Jtech Lasertool from Inkscape looks like this.
G1 Z0   

G90
G21
G1 F8000
G1 X58.2648 Y174.2597
G4 P0 
G1 Z1 S0
G4 P0
G1 F8000.000000
G1 X95.9498 Y174.2597
G1 X95.9498 Y131.2293
G1 X58.2648 Y131.2293
G1 X58.2648 Y174.2597
G4 P0 
G1 Z0   
G1 F8000
G1 X0 Y0

Was mache ich falsch?

Translated with www.DeepL.com/Translator (free version)

branther, 2021-06-04:
It is a pity that my request is not heeded.

gmmanonymus111, 2021-06-06:
When generating GCodes, you should have a laser power setting with at least `S1` after the `G0` command. Without that, LaserGRBL thinks your Gcode "graphics" is "empty". You need to generate gcodes a different way, or add `Sxxx` commands manually. See the differrence between:
```
M3 S0
S0 
G0X0.0Y0
S1
G1X50Y0F1000
Y50
X0
Y0
S0
G0X17.5Y17.5
S1
G1X25Y17.5F1000
Y25
X17.5
Y17.5
S0
M5 S0
```
and
```
M3 S0
S0 
G0X0.0Y0
G1X50Y0F1000
Y50
X0
Y0
S0
G0X17.5Y17.5
G1X25Y17.5F1000
Y25
X17.5
Y17.5
S0
M5 S0
```

Is there a particular reason you are generating GCode inside Inkscape rather than saving as a vector and importing that vector to LaserGRBL? That way you can use macros without messing with GCodes.

branther, 2021-06-07:
thank you for your support.
yes, there was a reason. namely, ignorance about the fact that lasergrbl can read vectors directly.
i tried it and it works very well.
unfortunately i don't use lasergrbl for lasering in this case.
I use GRBL-Servo-Pen, which does not react to M-codes.
But LaserGRBL unfortunately only generates M0 and M1 instead of Z0 and Z1.
Can I change this?

branther, 2021-06-07:
Sorry, I closed it by mistake.
yes, it now also works with g-code-generator from inkcsape out...
I had not used the commands M3 and M5.
I have entered them again and LaserGRBL now also recognises the dimensions.
Great. I am really very happy.
Thanks again for the support!

gmmanonymus111, 2021-06-07:
LaserGRBL currently does not support custom "laser on/off" commands, so you can't change it that way. One possibility is to save your LaserGRBL-generated GCode, then with notepad(++), replace all `M3/M4` laser on with `Z0` pen down and `M5` laser off with `Z1` pen up command (or anything you are using, I'm just assuming Z0 is down and Z1 is up)

branther, 2021-06-08:
Thank you, I use Z0 for up, i.e. switch off, and Z1 for down, i.e. switch on.
I have already been able to swap all the codes M3 and M5 for Z1 and Z0 using an editor (UltraEdit). This also works very well. It would be better if the fields in LaserGRBL were variable and not fixed to M3 and M5.
This would allow several variants to be processed with LaserGRBL.
best regards

closed 2021-06-20 by branther