arkypita/LaserGRBL #1461: MyPictureBox.cs

oliverhbailey, 2021-05-21:
Building lasergrbl using MonoDevelop yields generates the following error using 4.3.0 source code. That error indicates the sourc e file doesn't exist but it does and complte project clean was performed before the build.

/home/oliverhbailey/Downloads/LaserGRBL-4.3.0/LaserGRBL-4.3.0/LaserGRBL/CSC: Error CS2001: Source file '/home/oliverhbailey/Downloads/LaserGRBL-4.3.0/LaserGRBL-4.3.0/LaserGRBL/UserControls/MyPictureBox.designer.cs' could not be found. (CS2001) (LaserGRBL)

A clear and concise description of what the bug is.

**To Reproduce**
Download the lasergrbl-4.3.0.zip source code zip file in your home folder path
1. Open monoDevelop
2. Open the project file
3. Perform Clean All from the build menu
4. Perform Build all
5. Check Error

**Expected behavior**
A successful compile should have been performed.
**Screenshots**

If applicable, add screenshots to help explain your problem.

**Hardware and software configuration (please complete the following information):**
 - Debian 10 updated 6/21/2021
 - mono 6.12
 - monodevelop 7.8.4
 - Eleksmaker A3
 - Grbl version (i.e. 0.9j, 1.1h)
 - LaserGRBL 4.3.0

**Additional context**
It appears that the file MyPictureBox.designer.cs is missing from the source tree. It is referenced in the project, but missing from the source tree.

arkypita, 2021-05-21:
https://github.com/arkypita/LaserGRBL/tree/master/LaserGRBL/UserControls

repository show it is not missing. check case sensitivity

oliverhbailey, 2021-05-21:
It's not case sensitivity issue. System.Xml.Serialization is missing in
Mono 6.12. The assembly appears to be missing

This prevents it from building successfully on Linux.

oliverhbailey, 2021-05-24:
I looked further into the missing System.ml assembly. At the end of this comment are the 44 errors. There has been a lot of changes in Mono, Xmarin, and .Net since this was originallt built in 2017. Currently the .NET SDK is 5.0 and it is the consolidation of prior Windows (.NET), Linux (MONO), and Windows Forms, MaxOS, Android (Xmarin) into a single .NEt SDK.
System.XML Serialization is not in the .NET 5.0 SDK. But if you read the developer docs at Microsoft, Winforms will disappear completely in the next release .NET 6.0 coming in a few months. They mention specific System Assemblies that will be gone in the assemblies which require changes to ALL platform source code. It may be worth a read if you plan to continue compiling this product. The recommended changes will allow all C# 9 products to compile across all platforms once the changes are made. 

/home/oliverhbailey/Downloads/LaserGRBL-4.3.0/LaserGRBL-4.3.0/LaserGRBL/SvgLibrary/Paths/SvgPath.cs(18,18): Error CS0234: The type or namespace name 'Serialization' does not exist in the namespace 'System.Xml' (are you missing an assembly reference?) (CS0234) (LaserGRBL)
/home/oliverhbailey/Downloads/LaserGRBL-4.3.0/LaserGRBL-4.3.0/LaserGRBL/SvgLibrary/SvgElement.cs(41,41): Error CS0246: The type or namespace name 'XmlTextReader' could not be found (are you missing a using directive or an assembly reference?) (CS0246) (LaserGRBL)
/home/oliverhbailey/Downloads/LaserGRBL-4.3.0/LaserGRBL-4.3.0/LaserGRBL/SvgLibrary/SvgElement.cs(50,50): Error CS0246: The type or namespace name 'XmlTextWriter' could not be found (are you missing a using directive or an assembly reference?) (CS0246) (LaserGRBL)
/home/oliverhbailey/Downloads/LaserGRBL-4.3.0/LaserGRBL-4.3.0/LaserGRBL/SvgLibrary/SvgElement.cs(48,48): Error CS0246: The type or namespace name 'XmlTextWriter' could not be found (are you missing a using directive or an assembly reference?) (CS0246) (LaserGRBL)
/home/oliverhbailey/Downloads/LaserGRBL-4.3.0/LaserGRBL-4.3.0/LaserGRBL/SvgLibrary/SvgElement.cs(48,48): Error CS0246: The type or namespace name 'XmlTextWriter' could not be found (are you missing a using directive or an assembly reference?) (CS0246) (LaserGRBL)
/home/oliverhbailey/Downloads/LaserGRBL-4.3.0/LaserGRBL-4.3.0/LaserGRBL/SvgLibrary/SvgElement.cs(35,35): Error CS0246: The type or namespace name 'XmlTextWriter' could not be found (are you missing a using directive or an assembly reference?) (CS0246) (LaserGRBL)
/home/oliverhbailey/Downloads/LaserGRBL-4.3.0/LaserGRBL-4.3.0/LaserGRBL/SvgLibrary/SvgElement.cs(46,46): Error CS0246: The type or namespace name 'XmlTextWriter' could not be found (are you missing a using directive or an assembly reference?) (CS0246) (LaserGRBL)
/home/oliverhbailey/Downloads/LaserGRBL-4.3.0/LaserGRBL-4.3.0/LaserGRBL/PSHelper/MaterialDB.Designer.cs(25,25): Error CS0234: The type or namespace name 'Serialization' does not exist in the namespace 'System.Xml' (are you missing an assembly reference?) (CS0234) (LaserGRBL)
/home/oliverhbailey/Downloads/LaserGRBL-4.3.0/LaserGRBL-4.3.0/LaserGRBL/PSHelper/MaterialDB.Designer.cs(25,25): Error CS0234: The type or namespace name 'Serialization' does not exist in the namespace 'System.Xml' (are you missing an assembly reference?) (CS0234) (LaserGRBL)
/home/oliverhbailey/Downloads/LaserGRBL-4.3.0/LaserGRBL-4.3.0/LaserGRBL/PSHelper/MaterialDB.Designer.cs(72,72): Error CS0234: The type or namespace name 'XmlReader' does not exist in the namespace 'System.Xml' (are you missing an assembly reference?) (CS0234) (LaserGRBL)
/home/oliverhbailey/Downloads/LaserGRBL-4.3.0/LaserGRBL-4.3.0/LaserGRBL/PSHelper/MaterialDB.Designer.cs(54,54): Error CS0234: The type or namespace name 'XmlSchema' does not exist in the namespace 'System.Xml.Schema' (are you missing an assembly reference?) (CS0234) (LaserGRBL)
/home/oliverhbailey/Downloads/LaserGRBL-4.3.0/LaserGRBL-4.3.0/LaserGRBL/PSHelper/MaterialDB.Designer.cs(118,118): Error CS0234: The type or namespace name 'XmlSchemaSet' does not exist in the namespace 'System.Xml.Schema' (are you missing an assembly reference?) (CS0234) (LaserGRBL)
/home/oliverhbailey/Downloads/LaserGRBL-4.3.0/LaserGRBL-4.3.0/LaserGRBL/PSHelper/MaterialDB.Designer.cs(49,49): Error CS0234: The type or namespace name 'XmlSchemaComplexType' does not exist in the namespace 'System.Xml.Schema' (are you missing an assembly reference?) (CS0234) (LaserGRBL)
/home/oliverhbailey/Downloads/LaserGRBL-4.3.0/LaserGRBL-4.3.0/LaserGRBL/PSHelper/MaterialDB.Designer.cs(64,64): Error CS0012: The type 'XmlSchema' is defined in an assembly that is not referenced. You must add a reference to assembly 'System.Xml, Version=4.0.0.0, Culture=neutral, PublicKeyToken=b77a5c561934e089'. (CS0012) (LaserGRBL)
/home/oliverhbailey/Downloads/LaserGRBL-4.3.0/LaserGRBL-4.3.0/LaserGRBL/PSHelper/MaterialDB.Designer.cs(120,120): Error CS0234: The type or namespace name 'XmlSchemaSet' does not exist in the namespace 'System.Xml.Schema' (are you missing an assembly reference?) (CS0234) (LaserGRBL)
/home/oliverhbailey/Downloads/LaserGRBL-4.3.0/LaserGRBL-4.3.0/LaserGRBL/PSHelper/MaterialDB.Designer.cs(53,53): Error CS0234: The type or namespace name 'XmlSchemaComplexType' does not exist in the namespace 'System.Xml.Schema' (are you missing an assembly reference?) (CS0234) (LaserGRBL)
[... cut ...]
/home/oliverhbailey/Downloads/LaserGRBL-4.3.0/LaserGRBL-4.3.0/LaserGRBL/UserControls/MyPictureBox.cs(27,27): Error CS0115: 'MyPictureBox.OnMouseUp(MouseEventArgs)': no suitable method found to override (CS0115) (LaserGRBL)
/home/oliverhbailey/Downloads/LaserGRBL-4.3.0/LaserGRBL-4.3.0/LaserGRBL/UserControls/MyPictureBox.cs(27,27): Error CS0115: 'MyPictureBox.OnPaint(PaintEventArgs)': no suitable method found to override (CS0115) (LaserGRBL)
/home/oliverhbailey/Downloads/LaserGRBL-4.3.0/LaserGRBL-4.3.0/LaserGRBL/UserControls/MyPictureBox.cs(27,27): Error CS0115: 'MyPictureBox.OnPaintBackground(PaintEventArgs)': no suitable method found to override (CS0115) (LaserGRBL)

arkypita, 2021-06-21:
I am aware that microsoft is doing a lot of work to make .net cross-platform and that winform will be abandoned in the future, but this transition currently requires redesigning the entire user interface (not much in reality, but still heavy work) with very different paradigms (WPF etc) that are not immediately and quickly translatable from winform.

closed 2021-06-21 by arkypita