arkypita/LaserGRBL #1464: Adding support for 3d printers with lasers attached to the fan

albertoxamin, 2021-06-01:
**Is your feature request related to a problem? Please describe.**
I recently bought the reality laser that attaches to the carriage of 3d printers, unfortunately, the software available for this kind of laser is lacking.

LaserGRBL gcode cannot be used directly on those machines because the laser is controlled through the `M106 SXX.XX` command that is commonly used to control the part cooling fan of 3d printers.

I managed to create a script that converts the program output to usable gcode for this laser
```python
import re

file = open('input.nc', 'r')
lines = file.readlines()
out = open('output.gcode', 'w')

for l in lines:
    if 'M3' in l or 'M5' in l or 'M4' in l or 'S0\n' == l:
        out.writelines(['M107\n'])
    elif 'S' in l:
        s = re.findall(r"S[0-9]+", l)[0]
        out.writelines([f'M106 {s}\n', l.replace(f' {s}', '')+'\n'])
    else:
        out.writelines([l])

out.close()
```
Photo results of gcode produced with the script above
[image]
[image]
[image]

I know this is in the roadmap #64 but I thought this script could help someone until the feature is implemented.

**Describe the solution you'd like**
There are 2 possible solutions that I'm proposing:
1. Provide a way to specify post-processing scripts on the gcode that would be executed during the export. This feature would be similar to the one slicers for 3d printing offer
2. Add support for these kinds of machines in the dropdown where you choose the laser commands between M3 static or M4 dynamic

**Other stuff**
It might be related to #338 #784 #286 #274

closed 2021-06-21 by arkypita

7oRk3, 2022-07-13:
querido arkypita veo que marcas esto como completado. pero disculpame de verdad no e logrado encontrar el instalador como tal que contenga en la parte de importar una imagen que el codigo de prender el laser o apagarlo sean 106 y 107 si me pueden ayudar como tal con el .exe para ejecutarlo. porque no se como usar el scrit de [albertoxamina](https://github.com/albertoxamin) disculpen que no sepa mucho de esto

7oRk3, 2022-07-13:
@albertoxamin amigo buen dia. diculpa como ejecuto tu scrip. tambien pregunto habria una solucion nueva. que automaticamente me genere bien el .gcode para usar en la impresora 3d que prende el laser con m106 gracias

PaysenPetersen, 2023-02-09:
Hello, in what language is the script written?

albertoxamin, 2023-02-09:
it's python