arkypita/LaserGRBL #1466: machine moving in two axis

marcootaviomello, 2021-06-02:
I'm modifying a laser machine, which didn't use grbl firmware, used proprietary software along with proprietary company firmware. I recorded the GRBL on it, but when testing with the LASER GRBL when I click on the diagonal movement arrows, that is, it has command in both axes, it moves correctly, but when I click on the conventional up, down, left, right buttons, the same moves diagonally. ME seems to be something from mechanical construction, but can I solve it via GRBL configuration, to send pulse to both engines?

arkypita, 2021-06-02:
Picture of machine, controller, and mechanical parts please

marcootaviomello, 2021-06-03:
Analyzing the mechanics, this machine has only 2 motors, to make movements in a straight line I must send a command in both axes, (which is when we click on the arrow to move diagonally on the LASER GRBL), if I send a command only in one of the axes, the machine moves diagonally to the command axis. Any configuration ideas?
[image]
[image]
[image]
[image]

BeatSlayer, 2021-06-03:
That's a coreXY type of machine.

arkypita, 2021-06-03:
Exactly as @BeatSlayer say, it is a CoreXY
You need to flash a GRBL firmware compiled/configured for CoreXY (google it)

marcootaviomello, 2021-06-07:
Thanks a lot.