arkypita/LaserGRBL #1469: just stopped working??

madmike970, 2021-06-03:
cant export my config or modify it all of a sudden.... said to copy and paste this:
LaserGrbl v4.3.0.0
Grbl v1.1h
Wrapper: UsbSerial
Win 7, Home Premium, 6.1.7601.65536, Service Pack 1 (7)
CLR: C:\Windows\Microsoft.NET\Framework64\v4.0.30319\mscorlib.dll 4.8.4110.0 built by: NET48REL1LAST_B

TypeOf exception  [System.TimeoutException]
Exception message [Wrong number of config param found! (33/34)]
Exception source  [LaserGRBL], thread [Main Thread]
Exception method  [Void RefreshConfig()]

   ----------- stack trace -----------
   at LaserGRBL.GrblCore.RefreshConfig() in E:\LaserGRBL\LaserGRBL\Core\GrblCore.cs:line 789
   at LaserGRBL.GrblConfig.BtnExport_Click(Object sender, EventArgs e) in E:\LaserGRBL\LaserGRBL\GrblConfig.cs:line 178
   at System.Windows.Forms.Control.OnClick(EventArgs e)
   at System.Windows.Forms.Button.OnClick(EventArgs e)
   at System.Windows.Forms.Button.OnMouseUp(MouseEventArgs mevent)
   at System.Windows.Forms.Control.WmMouseUp(Message& m, MouseButtons button, Int32 clicks)
   at System.Windows.Forms.Control.WndProc(Message& m)
   at System.Windows.Forms.ButtonBase.WndProc(Message& m)
   at System.Windows.Forms.Button.WndProc(Message& m)
   at System.Windows.Forms.NativeWindow.Callback(IntPtr hWnd, Int32 msg, IntPtr wparam, IntPtr lparam)

mike
 norcolpromotion@gmail.com

gmmanonymus111, 2021-06-03:
It seems LaserGRBL failed to parse you GRBL `$$` config. It was looking for GRBL v1.1 config, which should be at least 34 parameters, but found only 33. (Migh be your settings got corrupted?)

Any more infos on your setup? (Machine, GRBL version etc)

madmike970, 2021-06-03:
this is the 2nd board it happened to.. once it says "cannot read config" in
the grbl config window its almost like it fries the arduino... even if i
upload the grbl master again and reinstall lasergrbl . but the arduino
isnt "fried" because i can still upload to it and get serial connection

gmmanonymus111, 2021-06-03:
AFAIK reflashing grbl will not clear your eeprom, where this data is stored. Try this in arduino: File->Examples->EEPROM->eeprom_clear before flashing grbl again

madmike970, 2021-06-03:
Thanks, I'll check it out.

madmike970, 2021-06-03:
THANK YOU SO MUCH!!!!  i thought i was frying boards or something!

madmike970, 2021-06-03:
so before i reinstalled everything i had it to where it would let me pick a
number between 0 and 1000 for laser power every time i loaded a file.. now
it just says m4 and m5 laser on off, how do i cfhange that back?

arkypita, 2021-06-21:
Just for information: "Wrong number of config param found" is an error that I explicit generate when reading configuration from the machine gives less config lines then expected.

GRBL v1.1 should have at least 34 parameters, [as documented here](https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration#grbl-settings)
GRBL v0.9 should have at least [31 parameters](https://github.com/grbl/grbl/wiki/Configuring-Grbl-v0.9#grbl-settings)
Older GRBL version [at least 23](https://github.com/grbl/grbl/wiki/Configuring-Grbl-v0.8)

If reading grbl config ($$) lasergrbl detect less then expected configuration, it is a clear indication of something wrong in flashed firmware (maybe a non custom firmware) so the configuration may not be interpreted correctly according to specifications.

closed 2021-06-21 by arkypita