arkypita/LaserGRBL #1473: Generated GCODE is not valid for Vectorize and other modes when device is set to marlin.

descipher, 2021-06-09:
**Describe the bug**
When processing any png with tools "1 bit dithering", "vectorize", "center line" the generated gcode contains the Snn parm on a new line.

**To Reproduce**
set device to marlin
open any usable png and gen the gcode 

**Expected behavior**
GCODE is processed line by line as per the standards, each line is a complete command.
The Snn parm must be on the same command line to be used for output power control of it associated prefix.
e.g. 
G1 x0.0 y0.0 S20

**Screenshots**
Centerline output clip:
G90 (use absolute coordinates)
M10 (Blower On)
M8 (Air Assist On)
X14.5Y365.3 **<- invalid** 
S0 **<- invalid** 
G0X37Y360 
S17 **<- invalid**
G1Y363.1F8000 

Line to line clip: **all valid**
G90 (use absolute coordinates)
M10 (Blower On)
M8 (Air Assist On)
G0 X0 Y360 F8000
M3 S0
G0 X2.188 Y360 S0
G1 X3.625 S16
G0 X6.75 Y360 S0 

Dither clip:
G90 (use absolute coordinates)
M10 (Blower On)
M8 (Air Assist On)
G0 X0 Y360 F8000
M3 S0
G1 X58.75 S17
S0 **<- invalid**
G0 X58.75 Y360.062 S0
G1 X0 S17
S0 **<- invalid** 
G0 X0 Y360.125 S0
G1 X58.75 S17

Vectorize clip:
G90 (use absolute coordinates)
M10 (Blower On)
M8 (Air Assist On)
M3 S0
F8000 **<- invalid**
G0 X2.446 Y360.5
S17 **<- invalid**
G1 X3.348 Y360.5
S0 **<- invalid**
G0 X3.351 Y360.667
S17 **<- invalid**

**Hardware and software configuration (please complete the following information):**
 - Windows 10 20H2
 - Custom w/Marlin 2.0.9.0
 - LaserGRBL 4.3.0

arkypita, 2021-06-21:
Hi, thanks for issue reporting.

I don't currently have time to devote to these changes, but I mark it as a bug so it stays active for when I can work on it.
Keep in mind that I can only test and develop LaserGRBL with GRBL fw, so it is very difficult to me to keep marlin code compatible.

descipher, 2021-06-22:
No worries, I may have a deeper look at the code if I find some time as well. If I do come up with something I will submit a PR to discuss it. Thanks for your work!

ctrysbita, 2021-09-08:
Any news here?

descipher, 2021-09-08:
Not at the moment, working on Marlin firmware laser cutter code right now. May look at this some time after that.

descipher, 2021-10-05:
I have resolved these issues within Marlin v2.0.9.2 we have a PR to be merged shortly that is compatible with LaserGRBL. Once merged you can simply enable #define GCODE_MOTION_MODES (remember GCODES) and issue GCODE->M3I to turn on inline mode then you can use LaserGRBL GCODE gens!

datibi, 2022-03-06:
Hi, I also faced these g-code issues with the current LaserGRBL version. Any hope to get a fix?

FarFlyer, 2022-10-03:
> I have resolved these issues within Marlin v2.0.9.2 we have a PR to be merged shortly that is compatible with LaserGRBL. Once merged you can simply enable #define GCODE_MOTION_MODES (remember GCODES) and issue GCODE->M3I to turn on inline mode then you can use LaserGRBL GCODE gens! There will be other issues but this is a big step forward.

Hello, I've Merlin version 2.0.9.2 with GCODE_MOTION_MODES enabled, but still I cannot make it working with S parameter on a new line.
Do you have any advice?

FarFlyer, 2022-10-06:
> The PR was merged in 2.1.x so you need to update it.

Hello, thanks for the reply. I've tried the version 2.1.1.

I have just one more hiccup: I need to modify the gcode manually changing the M3 command to M3I
And if I append more jobs, this has to be done for each job (so I have to find and replace all and I cannot simply add the M3I code at the beginning).

Is this the current situation also for you @descipher ?