arkypita/LaserGRBL #1499: hi

kre2, 2021-06-22:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Hardware and software configuration (please complete the following information):**
 - Windows version
 - Engraver brand and model
 - Grbl version (i.e. 0.9j, 1.1f)
 - LaserGRBL version

**Additional context**
Add any other context about the problem here.

gmmanonymus111, 2021-06-23:
This issue template is there for a reaon, please fill it according to your issue if it happened in LaserGRBL. If it's a general issue, and not connected to LaserGRBL, please open a discussion for it. No-one will be able to help you if you won't take the time explaining your issue.

gmmanonymus111, 2021-06-27:
Okay, I didn't get notification about your edit, only when the issue got closed.
Since your steps/mm settings correct (and LaserGRBL uses those settings to generate the gcode), I think there's a hardware issue somewhere. Is your psu strong enough to drive all the steppers along with three laser?
A picture about the issue could help here a lot.

kre2, 2021-07-01:
hi Alex
sorry for the late response....
laser use about 1.3A powered by 2.5A supply for laser only
printer have a 5A power supply for printer only (all steppers draw abot max
1.5A)
stepper drivers is tmc2100
i have very slightly pinched y axis while printing to see if that will
corse step skipping but no skipped steps
the steam logo was set to print at 70x70 mm (steam.png) but the outcome is
x=70mmy=64mm
itś not the only print with the same outcome
kind regards kresten

gmmanonymus111, 2021-07-01:
Could you attach the generated GCode here on Github? I'd like to do a test engraving to check what might go wrong. (If you can attach it, please do so on Github, as email attachments usually do not work :/)

Is the error margin the same ~6mm for all engrave jobs, is it almost 10% of the size or is it random?
When the job is finished, can your machine go to the exact `(0,0)` position where it started, or is that shifted as well?

kre2, 2021-07-13:
turns out no problem only my setup

closed 2021-07-13 by kre2