arkypita/LaserGRBL #1506: the laser does not turn off

benek1, 2021-07-04:
grbl ESP32
the laser remains on during pause and cooling.
There was no such problem when working with arduino. Is this program not compatible with ESP 32

gmmanonymus111, 2021-07-04:
This has nothing to do with LaserGRBL. Might be a GRBL firmware or hardware issue (especially, if it worked on an arduino).

benek1, 2021-07-05:
It remains for me to go back to arduino

arkypita, 2021-07-05:
Since grbl was originally designed for cnc machines, normally the spindle stay on even when the head does not move. With lasers this cannot be done because laser continues to burn hole.

[Official GRBL](https://github.com/gnea) firmware was developed for atmega328 (arduino) and since v1.1 there is a "[laser mode](https://github.com/gnea/grbl/wiki/Grbl-v1.1-Laser-Mode)" option (that should be enabled in order to work) which provides that every time the laser head stops moving the laser is turned off.

Other grbl versions for different microprocessor are done by third part, you should check if your version is based on v1.1 or not, if you have "laser mode" enable or not, and so if you have laser mode on but the laser stay on during pause and cooling you should/could contact firmware developer to open a bug report.

However this is not a LaserGRBL issue

benek1, 2021-07-05:
That explains it all. thanks

closed 2021-07-05 by benek1