arkypita/LaserGRBL #1509: Hi . When i connect my laser 3d printer ender3 with marlin 2.0fw ,it connect, show some lines from printer and then go off. Why i cannot get to stay running? what may be a problem?thx

etmipamidzi, 2021-07-05:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Hardware and software configuration (please complete the following information):**
 - Windows version
 - Engraver brand and model
 - Grbl version (i.e. 0.9j, 1.1f)
 - LaserGRBL version

**Additional context**
Add any other context about the problem here.

gmmanonymus111, 2021-07-05:
Please fill the issue template when opening an issue. It asks for info to help us solve the problem easier. If lest blank (like this), it does not contain any useful info for us to help you.

_Please note: Marlin support is experimental_
Did you set the firmware type to marlin? AFAIK, LaserGRBL defaults to GRBL, which sends status repots periodically to LaserGRBL (this way, LaserGRBL knows it is connected), but for marlin, I'm not sure this happens. (If it happends, I'm sure it's different, though). To change it, click to `Grbl` -> `Setings` -> `Firmware type`
[image]
[image]

etmipamidzi, 2021-07-05:
Thx but I cannot change M3 and M4 command for m106 and m107 for my laser.
How cai change it in new version?

gmmanonymus111, 2021-07-05:
Sadly, you can't right now. It's already in #64 as
- [ ] Customized code for laser on-of

You can, however, change it manually with a text editor: In LaserGRBL, click `File`->`Quick save`, chose a name & location, then edit the newly created file with notepad(++), and replace `M3` with `M106` and `M5` with `M107` (_depending on your config, of course_)

closed 2021-07-20 by arkypita