arkypita/LaserGRBL #1513: Error saving GRBL Configuration

HGeorgy, 2021-07-07:
LaserGrbl v4.3.0.0
Grbl v1.1z
Wrapper: UsbSerial
Win 10, Professional, 10.0.19042.0 (7)
CLR: C:\Windows\Microsoft.NET\Framework64\v4.0.30319\mscorlib.dll 4.8.4360.0 built by: NET48REL1LAST_C

TypeOf exception  [System.TimeoutException]
Exception message [Wrong number of config param found! (33/34)]
Exception source  [LaserGRBL], thread [Main Thread]
Exception method  [Void RefreshConfig()]

   ----------- stack trace -----------
   bei LaserGRBL.GrblCore.RefreshConfig() in E:\LaserGRBL\LaserGRBL\Core\GrblCore.cs:Zeile 789.
   bei LaserGRBL.GrblConfig.BtnExport_Click(Object sender, EventArgs e) in E:\LaserGRBL\LaserGRBL\GrblConfig.cs:Zeile 178.
   bei System.Windows.Forms.Control.OnClick(EventArgs e)
   bei System.Windows.Forms.Button.OnClick(EventArgs e)
   bei System.Windows.Forms.Button.OnMouseUp(MouseEventArgs mevent)
   bei System.Windows.Forms.Control.WmMouseUp(Message& m, MouseButtons button, Int32 clicks)
   bei System.Windows.Forms.Control.WndProc(Message& m)
   bei System.Windows.Forms.ButtonBase.WndProc(Message& m)
   bei System.Windows.Forms.Button.WndProc(Message& m)
   bei System.Windows.Forms.NativeWindow.Callback(IntPtr hWnd, Int32 msg, IntPtr wparam, IntPtr lparam)

HGeorgy, 2021-07-07:
Sorry, this error was emitted while trying to EXPORT the setting!
Regards

closed 2021-07-07 by HGeorgy

arkypita, 2021-07-07:
v1.1z is an unofficial version of grbl (controller board firmware). did you know where does it come from?

arkypita, 2021-07-07:
The error was emitted because your firmware is replying only with 33 config param while grbl v1.1 should have 34 params.

HGeorgy, 2021-07-07:
Hi arkypita,
thanks for your fast reply!
Honestly, I didn't even notice that there was a "z" appendix to the firmware! I downloaded nearly everything related to firmware which could possibly help to solve my "Homing"-problem! So what I read about the problem when you only have X and Y and the Laser is that the firmware triesto home Z first. This couldn't done with a laser device. Version  August 30th, 2019 seems to solve it, but not on my side. Still severe problems !

HGeorgy, 2021-07-07:
Related to your answer - 34 params! How can I solve this problem!?
Any idea would be welcome (....or a link where I can read about it)!

HGeorgy, 2021-07-10:
Hi arkypita,
reviewing my problem with "not able to saveGRBL config" - I counted through the parameters I can view with $$ and called up the list of configurations - there are always 34 parameters!!??
So why is this error emitted?

arkypita, 2021-07-10:
When reading grbl configuration LaserGRBL perform a background request of $$ configuration and capture all the stream received. Of course, being this a background operation, it should end at certain time.

This time is when an expected number of params are received and no more data are sent by controller, with a max timeout of 5 seconds. I after 5 seconds the correct number of params are not received lasergrbl throw an error.

[image]

Can you perform a session with extended log activated and upload your comlog.txt here?

- Open LaserGRBL
- Menu "?" activate extended log
- Chose where to save
- Connect the engraver
- Open GRBL configuration form
- Click on Export button

Please also upload "session log" (Menu "?" - "open session log")

HGeorgy, 2021-07-11:
Hi arkypita,
I performed the extended log and added the related error message as well - just in case.
One question: As I have also a svere problem with my Homing - should I open a seperate issue or can I discribe the problem here as well? Be so kind and let me know!
Regards
[comlog.txt](https://github.com/arkypita/LaserGRBL/files/6796741/comlog.txt)

sayego, 2021-07-11:
Please unsubscribe.

11 Tem 2021 Paz 13:24 tarihinde HGeorgy ***@***.***> şunu
yazdı:

HGeorgy, 2021-07-11:
What??

HGeorgy, 2021-07-11:
Sorry arkypita,
forgot the sessionlog!
[sessionlog.txt](https://github.com/arkypita/LaserGRBL/files/6796795/sessionlog.txt)