arkypita/LaserGRBL #1531: "LaserGRBL" software imports vector graphics

lianzaozi, 2021-07-18:
Can the "LaserGRBL" software add the function of importing vector graphics (svg dxf plt, etc.)? These files have the advantage that they have layer/color information, can identify layers or colors, and define different layers or colors Different laser power and speed.

StuartB4, 2021-07-18:
It does do .svg files.

lianzaozi, 2021-07-18:
that's good.

My question now is how to fill the shape in the svg file in the software?

arkypita, 2021-07-18:
you can use your favorite vector graphics program to create an infinite number of fills through the tools that these programs usually offer: patterns, strokes, clipping ...

https://graphicdesign.stackexchange.com/questions/45577/create-hatching-in-inkscape

closed 2021-07-20 by arkypita