arkypita/LaserGRBL #1546: Lasergrbl stop responding

Dhanyudi, 2021-08-04:
I am using only x and y axis.
Grbl 1.1h and Arduino uno.
But lasergrbl often showing error detected some problem with your board in the middle of progress.
Please help.
[image]

gmmanonymus111, 2021-08-04:
Have you read the [FAQ](lasergrbl.com/faq/) as mentioned in the error message?
If not, start with `“Problem detected” message: hang or disconnect during engraving` section.

For the Mega, I'm sure it is supported, as I'm using one myself (through telnet proxy though, but was working with USB before).
Try using `UsbSerial2` protocol in `Settings` -> `Connection protocol`
[image]

Dhanyudi, 2021-08-23:
> Have you read the [FAQ](https://lasergrbl.com/faq/) as mentioned in the error message?
> If not, start with `“Problem detected” message: hang or disconnect during engraving` section.
> 
> For the Mega, I'm sure it is supported, as I'm using one myself (through telnet proxy though, but was working with USB before).
> Try using `UsbSerial2` protocol in `Settings` -> `Connection protocol`
> [image]

When arduino mega flashing with xloader, it successfully connect with lasergrbl, but i cannot pass homing because using only 2 axis, how to disable z axis? Editing config.h and flashing with arduino ide cannot detect with lasergrbl

gmmanonymus111, 2021-09-05:
Sorry for the late reply, I've tried to make a tutorial how to edit & upload GRBL to a mega.
1. Get [GRBL Mega 5x](https://github.com/fra589/grbl-Mega-5X) (Click the green `code `button -> `download zip`)
2. Open the zip file, then extract `grbl` folder (only the grbl folder) to your arduino documents folder (default is `C:\Users\{YourUser}\Documents\Arduino\libraries`)
[image]
[image]
3. Once copied, you can edit `config.h` (Take a note about `#define BAUD_RATE` ardound line 48, you need to use this number when connecting with LaserGRBL)
4. Open Arduino IDE.
5. Click `File` -> `Examples` -> `grbl` -> `grblUpload`
[image]
6. Select the correct board & COM port
[image]
7. Hit upload (once uploaded, you should get similar orange text)
[image]
8. Connect with LaserGRBL (You can use `UsbSerial2 `protocol, that seems to be more stable with mega)
[image]
[image]
I hope this tutorial helps

Dhanyudi, 2021-09-05:
> Sorry for the late reply, I've tried to make a tutorial how to edit & upload GRBL to a mega.
>
> 1. Get [GRBL Mega 5x](https://github.com/fra589/grbl-Mega-5X) (Click the green `code` button -> [...]

Wow...thank you very much for detail tutorial.
I will try , and send further info

Dhanyudi, 2021-09-06:
I have try and still no luck
I can not connect to the mega with lasergrbl

gmmanonymus111, 2021-09-06:
> I try and still no luck
> I can not connect to the mega with lasergrbl

Does LaserGRBL show any error messages?
Could you upload LaserGRBL extended log after an unsuccessful connection? 
Could you take pictures when UGS connects?

Dhanyudi, 2021-09-06:
[image]
[image]

Dhanyudi, 2021-09-06:
[image]
Error

Dhanyudi, 2021-09-06:
[image]
Log

Dhanyudi, 2021-09-06:
[image]
Using ugs port not detect, but in device manager there is a port arduino

Dhanyudi, 2021-09-06:
[image]

Using ugs port not detect, but im device manager port is detect arduino

gmmanonymus111, 2021-09-06:
I suspect another issue here, since neither LaserGRBL nor UGS can see your MCU. 
By extended log, I meant `?` sign in LaserGRBL -> `Activate extended log` -> Select a path where you want to save it -> Try to connect -> When failed, you can upload the log here, it might contain something useful info.

What kind of mega do you have? The original/clone with ATmega16U2 (big brick like connector), or CH340 micro USB connector? CH340 is known for not being that reliable.

When finished flashing the firmware (in Arduino IDE), when opening `Serial Monitor`, do you see a message similar to this?:
[image]

Dhanyudi, 2021-09-06:
> I suspect another issue here, since neither LaserGRBL nor UGS can see your MCU.
> By extended log, I meant `?` sign in LaserGRBL -> `Activate extended log` -> Select a path where you want to save it [...]

Nope i dont see any message in serial monitor

gmmanonymus111, 2021-09-07:
What is the output text when you finished flasshing in Arduino IDE? (the orange text at the end)
I suspect either your USB cable, USB port or Arduino USB to serial converter might be damaged, since there is no reliable communication between your pc and your board