arkypita/LaserGRBL #1590: 4.4.1 bad curve vs good 4.4.1 (svg)

edsonee, 2021-09-28:
**Describe the bug**
hello, version 4.5.1 does not generate the curves well, I send some photos.
version 4.4.1 if I did them right.
[image]

grbl 1.1F
machine generica china
windows 10

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Hardware and software configuration (please complete the following information):**
 - Windows version 10
 - Engraver brand and model
 - Grbl version 1.1f
 - LaserGRBL version 451

**Additional context**
Add any other context about the problem here.

arkypita, 2021-09-28:
It's something I've actually changed in the latest version and it was supposed to bring better results, but it doesn't seem to be (at least in this case). Details here: https://github.com/arkypita/LaserGRBL/pull/1543

Can you share the original .svg so I can test with it? I can try to find a better implementation or if I can't I revert this change back.

arkypita, 2021-09-28:
If SVG file format is not allowed to upload, just zip it and upload zipped file.

edsonee, 2021-09-28:
I send original .svg

thank you

atte Edson

edsonee, 2021-09-28:
svg zip

SIP...[http://gfx1.hotmail.com/mail/w4/pr04/ltr/emo/winking_smile.gif]!!!

edsonee, 2021-09-28:
[logo honda.zip](https://github.com/arkypita/LaserGRBL/files/7247496/logo.honda.zip)

logo svg zip

edsonee, 2021-10-01:
Arkypita  please read #1591 is a suggestion not a question

https://github.com/arkypita/LaserGRBL/issues/1591
[https://opengraph.githubassets.com/2c052d5a8d126eadfcca6a4f9d378bbd0a7a6fe2ec6f7c2f1cb80a253e0855d9/arkypita/LaserGRBL/issues/1591]<https://github.com/arkypita/LaserGRBL/issues/1591>
suggestion to : Why so little power if it is at 100? What is Smax? Why 1000? · Issue #1591 · arkypita/LaserGRBL<https://github.com/arkypita/LaserGRBL/issues/1591>
Hello friends, first of all, I thank the creator of this great program. This suggestion is based on a frequently asked question from new users. (I belong to a group of lasergrbl whatsapp and it alw...
github.com

SIP...[http://gfx1.hotmail.com/mail/w4/pr04/ltr/emo/winking_smile.gif]!!!

arkypita, 2021-10-05:
Fixed in v4.6.0 currently pre-release, can be downloaded here for test: https://github.com/arkypita/LaserGRBL/releases/download/v4.6.0/install.exe