arkypita/LaserGRBL #1591: repeated question. Why so little power if it is at 100? What is Smax? Why 1000?

edsonee, 2021-09-30:
Hello friends, first of all, I thank the creator of this great program. This suggestion is based on a frequently asked question from new users. (I belong to a group of lasergrbl whatsapp and it always asks the same thing)
Why so little power if it is at 100? What is Smax? Why 1000? 

[image]
[image]
[image]

I suggest SMAX work on percentage and name change. (see picture 3)
It is a suggestion designed for the comfort of the user
I know that everything is explained and documented on the same page and old users have no problem with this issue. but new users are always confused.
thank you very much.

**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

arkypita, 2021-10-01:
https://lasergrbl.com/wp-content/uploads/2020/05/Grbl-Configuration-ENG.pdf

edsonee, 2021-10-01:
I know that everything is explained and documented on the same page and old users have no problem with this issue. but new users are always confused.
thank you very much.
I suggest SMAX work on percentage and name change. (see picture 3)
It is a suggestion designed for the comfort of the user

https://lasergrbl.com/wp-content/uploads/2020/05/Grbl-Configuration-ENG.pdf

edsonee, 2021-10-01:
[image]
[image]
[image]

alastaira, 2021-10-02:
@edsonee g-code - the language which LaserGRBL (and all GRBL controllers use), was originally designed to control spindles on CNC machines. As such, the "S-Max" value referred to the maximum rotation speed of the spindle. One spindle might rotate at a max of 10,000RPM, while another had a maximum of 20,000RPM, so the $30 value was designed so that users could specify that, and ensure they wouldn't exceed the maximum.

A speed of 10,000, on a machine capable of spinning at 20,000RPM, would be 50%.
A speed of 10,000 on a machine capable of spinning at 10,000RPM, would be 100%.
But they'd be rotating at the same speed..... so it makes no sense to express as a percentage.

The concept of "rotation speed" doesn't really apply to lasers, so people generally choose either 255 or 1000 as their "maximum rotation speed", and then specify some fraction of that. It's not a LaserGRBL issue - it's just how gcode has been developed.

edsonee, 2021-10-02:
Thanks for answering, I know that gcode in its programming does not work with percentage ($ 30) but for new users it is easier to express the power of the laser in%, a new user will never say "I want to work at 255 of my laser "or" I want to work at 1000 of my laser ". They don't know what "smax" or "$ 30" is. That is why I suggest that the interface works in% and that internally LASERGRBL does the transformation. example (100% = $ 30 255) or (50% = $ 30 255/2).
additionally you give examples of RMP but LASERGRBL is designed to work with laser, and users tend to say "I want to work at 100% of the power of my laser".

Thank you for reading my comment.

SIP...[http://gfx1.hotmail.com/mail/w4/pr04/ltr/emo/winking_smile.gif]!!!

alastaira, 2021-10-02:
@edsonee LaserGRBL already shows the calculated percentage values of the supplied values as shown in the screenshot above, but you want the option to do it the other way around? (i.e. users type a percentage, and it converts to a value?). I suppose I could see a possible value in that, but I wonder if it would make the interface confusing to allow both methods.

edsonee, 2021-10-02:
alastaira you say: you want the option to do it the other way around? (i.e. users type a percentage, and it converts to a value?)
YES, that users enter percentage and next to this the value (proportional, see image)
Thanks Alastaria!
(My native language is not English, maybe I don't express ideas well).

If you think of it as a programmer this modification is unnecessary but if you think of it as someone who is going to use LASERGRBL for the first time, it will help him.

edsonee, 2021-10-02:
MENU SVG
[image]

arkypita, 2021-10-03:
> MENU SVG [image]

Hi

For consistency I should add percentage also in SVG menu.

M3 is necessary (you can select M3/M4 for SVG too). M5 it is always superfluous since there is no other option, but always for the same reason does not bother.

Regarding the request to modify the UI to work directly in percentage I reply that it is something that I had assumed but that I have discarded because, working in percentage, requires the assumption that the GRBL configuration values ($30, $31) are always known. 

This is not always true!

For example until you connect a machine for the first time, for example when you use LaserGRBL to generate GCODE for a machine connected to another system (for example because it is not a GRBL machine), for example for those firmware (marlin, smoothie) for which there is experimental support (perhaps not very functional) but which do not plan to have the parameters $ 30 and $ 31.

For all these situations the choice to use the S-Value is much more correct and opens up more possibilities to use a percentage value.

closed 2021-10-05 by arkypita