arkypita/LaserGRBL #1594: Nasty bug in G3 arc coordinates created from vectorisation of PNG file - possible safety hazard!

alastaira, 2021-10-02:
Selecting the Vectorize function for a particular PNG image (attached) generates a set of instructions containing a G3 command with extreme I and J parameters that cause the laser to move massively out of range (while burning!). That instruction is not shown in the preview window and there is no indication of a problem until LaserGRBL tries to execute that command.

**To Reproduce**
1. File Open the attached file "wym_animalcipher.png"
2. Choose settings as shown in "LaserGRBL_Bug.png" screenshot - vectorize, mid-level brightness/contrast/white clip, B&W, spot removal 2.0, Smoothing 1.0 Optimize 0.2, No filling. 
3. On Target Image dialog, choose Autosize to select 300 DPI (I'm using border speed 1800, if it matters)
4. Note that, in the preview window, there is a slight break on one side of the letter "n"
5. Start the job. After only a few seconds executing, the command highlighted in "LaserGRBL_Bug2.png" will be executed, causing the laser to move to extreme positive y value (I don't know the exact distance as I killed the job as soon as it started!)

**Expected behavior**
Correct vectorisation of the image (and/or correct visualisation of the preview of the commands generated, so at least the error would become apparent before starting the job)

**Hardware and software configuration (please complete the following information):**
 - Windows 10 64-bit
 - Eleksmaker Mana SE 3.2 controller
 - Grbl version v1.1f
 - Reproduced using both LaserGRBL v4.2.1 and v4.4.
[image]
1
[image]
[image]

alastaira, 2021-10-02:
It seems likely that this shares the same cause as #1248

arkypita, 2021-10-03:
Hi, can you share the original wym_animalcipher.png file put inside a .zip? The file attacched in github is compressed.

alastaira, 2021-10-03:
Sure. Please find attached (lines highlighted on above screenshot occur on Lines 161-162 of the .nc file)
[wym_animalcipher.zip](https://github.com/arkypita/LaserGRBL/files/7273785/wym_animalcipher.zip)
[wym_animalcipher.nc..zip](https://github.com/arkypita/LaserGRBL/files/7273800/wym_animalcipher.nc.zip)
.

arkypita, 2021-10-03:
Hi, I tried to send the code to my marker and I confirm the wrong (and dangerous) behavior.

However, the code generated by LaserGRBL is not entirely wrong. The I and J values are the offset values of the center point of the arc, relative to the starting point of the circle. It is completely normal that I and J are outside the work area and it is also normal that they can assume quite large values when working with short and almost straight arcs.

[image]

If you upload the gcode file you sent into a simulation program like [CAMotics](https://camotics.org/download.html) you can see that a PC has no problem simulating this path.

I believe that the issue is inside the grbl firmware instead, that could be not able to handle this numbers (hardware limit).
See: https://github.com/gnea/grbl/issues/901

arkypita, 2021-10-03:
Since an arc of circumference with a very distant center is actually approximable with a straight line, I think I will try to intercept these cases and transform them into straight lines.

alastaira, 2021-10-03:
That sounds like a reasonable fix although, as you pointed out, it might only be covering up a fault in GRBL itself. Do you think I should raise it in the issue queue there as well? (I don't think new features are being added to GRBL, but I'm not sure about bug fixes).

alastaira, 2021-10-03:
Thankyou for your attention. I agree - it seems likely that this is caused by the GRBL issue you referenced concerning large I/J values parsed as single precision floats.
It would be nice if the underlying cause were addressed (i.e. GRBL itself sanitised any inputs) but, as an immediate fix, I think your suggestion of creating straight line segments instead of arcs when the deviation is so minor is a good alternative.

closed 2021-10-03 by arkypita

arkypita, 2021-10-05:
Implemented in v4.6.0 currently pre-release, can be downloaded here for test: https://github.com/arkypita/LaserGRBL/releases/download/v4.6.0/install.exe

commit 5f4e5a97 (2021-10-03): Fix #1594 G2/G3 with huge I, J