arkypita/LaserGRBL #1598: SW does not end at the end

ronisegoly, 2021-10-04:
I use lasergrbl to create gcode from image
recently I noticed the SW does not end at the end of the process, it stops at last point, no end message, and I have to stop it manually.
https://www.dropbox.com/s/7ltu9usafgtldga/portrait_Iwkfs_IMG_3970.nc?dl=0
https://www.dropbox.com/s/pj5g33trdhpv2s6/portrait_Iwkfs_IMG_3970%20%281%29.JPG?dl=0

arkypita, 2021-10-04:
If work for me.

[image]

Note that the program ends when lasergrbl receives the last "ok" response from the controller at the last command sent. If there is no answer, the program is not finished. In this case you see the command with a "clock" near it.

ronisegoly, 2021-10-04:
So the issue is on my HW side. I ran now very short code, sometime I hear beep at the end, the head goes back to 0,0 and all status are green V. And sometime I see clock mark at the end, no V and it does not return to 0,0
Will talk to my HW support

arkypita, 2021-10-04:
Well, the [protocol between LaserGRBL and the machine](https://github.com/gnea/grbl/wiki/Grbl-v1.1-Interface) provides that for each command sent the machine responds with a reply message which can be "ok" or "error". This message arrives when the machine has read and accepted (validated) the command and warns the PC that it is possible to send a new command (if available).

In reality it is also possible to send more than one command, because there is a small reception buffer in the machine, but it is still necessary for the machine to respond with an acknowledgment for each of the queued commands.

If this "ok" message doesn't come, it's a problem.

Normally the machine always sends an "ok" message, the problem is that sometimes it is "lost" on the cable due to electrical noise, or it is merged with other messages that the machine sends (for example the message that carries current head position).

LaserGRBL contains some mechanisms (and I am trying to add more) to detect this situation, but it is a problem that must be solved at the source because it can hide others.

ronisegoly, 2021-10-04:
I need to investigate
I use lasergrbl for long time, not sure when the issue first spotted, I think when I bought a new CNC machine
I have now two machines connected to my PC.
On the CNC I spotted issues with PC machine communication, because is stopped progressing, I followed advice from supplier (disabled USB selective suspension on PC) and also reduced load by using lower speed for carving, it never appeared again.
On lasergrbl side I noticed several times recently, that the process freezes and then resume, did not check the screen at the time, but what I described now happens always at the end of the session.
I will investigate

ronisegoly, 2021-10-04:
what is
04/10/2021 08:22:36.689 Issue detector Handle Missing OK [11]
04/10/2021 08:30:23.289 Issue detector Handle Missing OK [8]

arkypita, 2021-10-04:
This is just one of those mechanisms I mentioned earlier. When these "ok's" are lost LaserGRBL would at some point stop sending new commands, because it believes there is no more space in the receive buffer. 

[image]

LaserGRBL performs a continuous check to understand if we are in this situation (he knows that for him the buffer is full, he understands that the machine is not moving, the machine is telling him that the buffer is empty) and in this case it self-simulates a series of "ok" to balance the missing ones.

Without this mechanism you would see the machine stop at some point with a series of commands with the clock.

arkypita, 2021-10-04:
You can find some hint under https://lasergrbl.com/faq/ section “Problem detected” message: hang or disconnect during engraving

arkypita, 2021-10-05:
Some improvement in v4.6.0 currently pre-release, can be downloaded here for test: https://github.com/arkypita/LaserGRBL/releases/download/v4.6.0/install.exe

ronisegoly, 2021-10-07:
I did nothing yet and did not encounter the issues again
I installed the pre release, that's the log
https://www.dropbox.com/s/x2hzwa218p5lfqn/sessionlog.txt?dl=0

arkypita, 2021-10-07:
What engraver and controller board do you have?
What values do you have here?

[image]

closed 2021-10-13 by arkypita