arkypita/LaserGRBL #1599: Smoothieware Connection via telnet port 23 disconnection

augustenlee, 2021-10-05:
Hi sir,

Good day, and I am so glad to have this change to report you this:
I got the latest Smoothieware firmware (using CNC mode) and try to make connection to board via telnet(port 23).
Once I button down the "connect" , but it disconnected immediately.
after I read the log file , there was something strange while system parsing the MACHINE POSITION.
attached pictures for you that you will understand what my question is ...

the response from smoothie ware board is: 
[image]

and the SmoothieCore.cs is:

[image]

I want to ask that if I can make a little modification of the code if you agree it ,, like it:
[image]

thank you very much!

augustenlee@gmail.com

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Hardware and software configuration (please complete the following information):**
 - Windows version
 - Engraver brand and model
 - Grbl version (i.e. 0.9j, 1.1f)
 - LaserGRBL version

**Additional context**
Add any other context about the problem here.

arkypita, 2021-10-05:
The code of the ParseF function is correct as it acts only on the F: 400.0,102.0 part of the string.
There is no reason to change it.

> after I read the log file , there was something strange while system parsing the MACHINE POSITION.

Post the error message you find in the log file.

augustenlee, 2021-10-05:
[sessionlog.txt](https://github.com/arkypita/LaserGRBL/files/7287166/sessionlog.txt)
Hi,sir
attaching log for you...
thanks!

augustenlee, 2021-10-05:
I think maybe the reason is the CNC mode of the firmware..

augustenlee, 2021-10-05:
Okay, I will try to parsing the "F:400.0,102.0"
thank you very much..

augustenlee, 2021-10-05:
Hi, Sir,
thank you very much,
I thought that I found the problem already.
[image]
.

and the answer from my machine is :
<Idle|MPos:0.0000,0.0000,0.0000|WPos:0.0000,0.0000,0.0000|F:400.0,102.0>

arkypita, 2021-10-05:
The "`<Idle,MPos:0.000,0.000,0.000,WPos:0.000,0.000,0.000>`" is the code to handle status report from grbl version <= 0.9

If your firmware reply with <Idle|MPos:0.0000,0.0000,0.0000|WPos:0.0000,0.0000,0.0000|F:400.0,102.0> it should be a v1.1

What grbl version is printed out when you connect the engraver?

arkypita, 2021-10-05:
I believe that your LaserGRBL never receive a grbl "welcome message" so he doesn't know what version is it. This is how welcome message look like

[image]

augustenlee, 2021-10-06:
> I believe that your LaserGRBL never receive a grbl "welcome message" so he doesn't know what version is it. This is how welcome message look like
> 
> [image]

No, I haven't seen this information since my connection type was set to "Smoothie"
but thanks any way,
I will try to connect this night..

Sincerely,

Augusten

arkypita, 2021-10-06:
I believe that since there is a telnet channel involved, grbl sends its welcome message but this is lost because the PC is not connected yet. Maybe you can force re-sending a welcome message by click on thunder symbol (I don't know if smoothie reply to this command with welcome message).

If it doesn't work you can do this: set Type = GRBL and Protocol Emulator, connect once to the emulator so LaserGRBL sees the message Grbl v1.1 and configures itself for that version. Once he has seen it he should remember it.

augustenlee, 2021-10-10:
dear Sir, r
after checking my smoothieware’s grbl version, It was not V0.9..
 i modified the code  to decode the smoothie gcode as V1.1e
then it worked!

thank you very much…