arkypita/LaserGRBL #1620: vector mode  settings

wjhutton, 2021-10-17:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

on some vector images that i burn the image is destroyed when it does the the outline border.
in the vector generation dialog
You have parameters  for border speed and filling speed, but only one laser strength setting Smin and Smax.
This becomes an issue with text in logos, when the logo or text is very small. It is also a problem with a complex image that has very fine (thin) edges.

Second minor issue consider adding a check box, or set the default to return to x0 y0 on job completion

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

please consider
 - adding a  Check box,  default checked:  to enable/disable border  code generation
- secondly add another Smin and Smax  for border

This is an excellent product, and works very well, Thank you for all of your work.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

what i currently do is set different speeds, Save the .nc file, open in an editor and locate the border/outline
block and either delete the block, or global search and replace to set a new Smax and if necessary adjust the speed

on some occasions i have removed sections of the outline where it has a tendency to over burn edges.

**Additional context**
Add any other context or screenshots about the feature request here.

wjhutton, 2021-10-18:
> Thinking this through further
> Suggest placing Smin Smax for fill and border side by side with a title
> for each column in laser Options
>
> LASER OPTIONS
> Laser Mode .......................................
> Fill    "fill mode"           Border   enabled [  ]
> S-MIN [___]  100%    S-MIN [___]  100%
> S-MAX [___] 100%    S-MAX [___]  100%
> Warning text line located here to explain the users choice of no output,
> or to explain why they cannot uncheck "border enabled"
> return to WCS X0 Y0 [checked] on job completion
>
> "fill mode" can be "enabled" or "none" or a repeat of selected option or
> empty if a mode selected.
> if no fill selected  the column could be shaded out disabling any input
> with settings of 0
> likewise if border enabled unchecked this column could be shaded with
> settings of 0

final thought and i believe the best option, requires display of one of two
dialog boxes for laser settings

When clicking on next on import raster image box  - check fill selection -
if none  open "target image border" dialog
this box only has border fields with no ability to disable the border

if fill selection not "none"   open "target image" dialog
this box has fields for both fill and border, with an option to skip the
border process, and a check box to indicate that it is a cut
(cut option would ensure inside cuts are made before the perimeter)

this solves the issue of no output, and eliminates any confusion on
parameters used

i still like the program to position at wcs x0 y0   so that another file
can be loaded to cutout the material ensuring it starts at the correct
location

i have been generating files with an image line by line, followed by a
vector file, followed by a cut file.

thanks for considering this. I hope this input helps you

PrinceNelson94, 2025-01-06:
This would be a really useful feature. I've also found the the border can destroy fine details. Any thoughts on implementing this?