arkypita/LaserGRBL #1628: LaserGrbl - laser movement by double-clicking does not work when the soft limit is switched on. ($ 20 = 1)

igortv1, 2021-10-19:
**Describe the bug**
when the soft limit is enabled, double-clicking on the selected XY location does not work. the laser start run up at the top right and stops by hard limit. if the soft limit is switch off, everything works as it should and the laser moves to the given location.
I think it's a SW bug. 

**To Reproduce**
just double-click on a specific location of the lasergrbl area.

**Expected behavior**
double-clicking on the XY position should move the laser to that position, even when the laser mode is switched on.

**Hardware and software configuration (please complete the following information):**
 - Windows version W7 ultimate 64bit
 - Engraver brand and model DIY CNC3018
 - Grbl version 1.1f
 - LaserGRBL version 4.6.1

**Additional context**
Hello, I'm here for the first time and forgive me for not knowing where to turn with the problem.
Thank you for understanding
Igortv1.

[config_grbl_009-SL_ena.txt](https://github.com/arkypita/LaserGRBL/files/7375265/config_grbl_009-SL_ena.txt)

arkypita, 2021-10-19:
Thanks for the detailed report, I will check the situation as soon as possible.

arkypita, 2021-10-20:
I have tested with my OLM1 with both homing and soft limit, as per your config, and it work for me.
Are you sure that it is not an issue of your engraver firmware? 

Here a video: https://youtu.be/Z-Bvr9ltBNE

igortv1, 2021-10-21:
Thank you for your time and video, I haven't flashed FW even once. when there is an error, then both the machine and the cursor move identically. therefore, I assume that the machine will be OK. it occurred to me that there could still be an error in setting the directions of the XY axis ($ 3, $ 23). I'm going to check it out.

igortv1, 2022-03-30:
link on video is there:
https://ulozto.cz/tamhle/ycXSy7LXYE2n#!ZJD4AwR1MzLkMGN4LGt5MGtkAwp3owWSJJ5UK1OXqySbA2L1

StuartB4, 2022-03-30:
Are you using the offline controller at the same time as the usb is connected. I know them 3018 machines don't like it if both are connected.

StuartB4, 2022-03-30:
Can't see the video, won't play and won't let me download without signing up.

igortv1, 2022-03-30:
please click on right button download free but slow
[image]
.

StuartB4, 2022-03-30:
Ok, got the video. Noticed when you click connect the cordinates in the top right of the main window are not X: 0.000 and Y: 0.000.
Like this. 

[image]

I have these both set and LaserGRBL always starts with X: 0.000 and Y: 0.000 when I connect. 

[image]

igortv1, 2022-03-30:
i have this the same, except hard reset.
i am enabled hard reset, and situation was changed. after doubleclick now don't any move and written "travel exceeded
[image]
"

igortv1, 2022-03-30:
of course with $20=1. with $20=0 everything is still OK

arkypita, 2022-04-01:
[image]

Please read what's here inside, this is the command LaserGRBL send.

$J=G90 is a Jog command using absolute coordinates.
Commanded position is X30 Y42, speed F1000

The travel exceeded is generated as an error by the Grbl controller, as response to this command.
You should try to understand why the macine is refusing this positioning command.

If you do jog using arrows, it works or not?

I think you could have a better help on [grbl firmware github page](https://github.com/gnea/grbl/).

Herbi5, 2022-12-10:
Hi, I am facing the same thing. When double clicking the canvas (Engraving preview) the laser head just continues passing the click point and stops at hard/soft limits. I noticed that in absolute position command ($J=G90X1502.00Y1502.00F1000) generated by double click, there are coordinates of X-axis maximum travel $130 and Y-axis maximum travel $131 in my case 1500 mm + 2 mm for Homing switch pull-off distance. My homing position is at upper right corner where end switches are located and then I am pulling back X-1500 Y-1500 (negative machine space) where I set zero point (G92) --> physical left, bottom corner. When jogging with arrows keys setting the relative coordinates (G91) everything works well. Behavior is same as igortv1 observes and as shows in his first video: https://ulozto.cz/tamhle/bAiFzDIubVoA#!ZGLlZQR1LzEzMQEuZQLmLzH3ZGD5DJ9JLxZko0kxYGSdBGp5.

Here too, one can notice upon clicking the value set by LaserGRBL is $J=G90X287.50Y177.00F1000 (287.50 and 177.00 are his maximum travel values according to his config.txt presented here) instead setting it to cca. $J=G90X30.00Y40.00F1000 where the click on the canvas appeared! --> Seams to be some bug in code of LaserGRBL when clicking on the canvas with set soft limits and negative machine workspace.
[image]
[image]
[image]

Grbl version 1.1h
LaserGRBL version 4.8.0

arkypita, 2022-12-11:
I tested what you reported. Again, with an official GRBL firmware, the behavior is correct with no issues. The problem you are reporting is probably caused by incorrect implementation of a custom firmware.

https://user-images.githubusercontent.com/8782035/206917253-b9531879-c15c-4ebf-8e82-3a986a5cb694.mp4

arkypita, 2022-12-11:
Ok, I have found what point of the code change the clicked coordinates to the maximum travel coordinates.
Is it a safety feature that try to limits the jog command, to the maximum machine travel.

[image]

arkypita, 2022-12-11:
Ok, you are right.
It does not happens to me because I don't have soft limits, if I enable them I have the same effect.

arkypita, 2022-12-11:
Apparently, the code that limit jog to soft limits seem to be too complicated.

Math.Min(Math.Max(target.X, -mWCO.X), (float)Configuration.TableWidth - mWCO.X);
Math.Min(Math.Max(target.Y, -mWCO.Y), (float)Configuration.TableHeight - mWCO.Y);

can be simplified to 

Math.Min(target.X, (float)Configuration.TableWidth); 
Math.Min(target.Y, (float)Configuration.TableHeight); 

But I have to think about possible side effects

Herbi5, 2022-12-11:
I was looking your sources yesterday evening and came to the same lines of code but I could not debug it since I currently have no access the to the machine. I was just debugging using emulator but there the soft limits doesn't work or are not implemented (do not report error if they are overrun). When I get machine to machine I will debug this part so that will suit my    needs, unless you will already introduce some modification. 
O I see you are working already on it. 

Good job tnx.

arkypita, 2022-12-11:
if you are able to build from source code, please test by yourself and let me know.
I have no access to a machine now (I am using a arduino board, but no hardware connected)

Herbi5, 2022-12-11:
I will test this end of the week, since the machine is at my sister. I am modifying it for her to do the job what she wants (large draw plotter). 

So I should try your simplification:

Math.Min(target.X, (float)Configuration.TableWidth);
Math.Min(target.Y, (float)Configuration.TableHeight);

[image]

I will do that and I will let you know the results in a week :)

Regards,

Herbi5, 2022-12-14:
Hi, I managed to test your simplification and it works as expected. This was short test, so I could not identify if this implementation somehow influences execution in other sections of the code (side effect you mentioned).

https://user-images.githubusercontent.com/120283386/207718151-c8964d69-50c2-4dbf-8b47-2296b5b6c0af.mp4

Thanks for the proposed solution and keep up the good work. I might some day encourage myself and translate the GUI into Slovene. ;)

igortv1, 2022-12-31:
Thank you also for clarifying the problem and fingers crossed. Also happy New Year's Eve and many good attacks in new year 2023.