arkypita/LaserGRBL #1631: lasergrbl setup simplified

philparvin, 2021-10-29:
I understand the basic function of the gcode BUT
it seems to me that there are some basic things that need to be understood...
the program moves the head to the locations where burn is to be
the laser is turned on at whatever intensity set in code
and my question is, is there not a length of time that the laser burns on each point?... what is that setting
In other controller software, I can set that variable so the laser doesnt have to be set at 100%, it just burns longer at each point which seems to make a sharper product.
Am I missing the mark ? is there a gcode setting for the length of burn time on each point?
thanks

sergejzr, 2021-10-31:
Hi  @philparvin ,
I am also new to all this, trying to write simple scripts to convert multilayered SVG into multistep GCode. 
I believe, the parameter you need is called "Feed Rate" and it is a the velocity mm/min the laser (or spindle) moves. You set it with "F"
Example:
G1 X10.0  Y20.0 F600
means "move to 10/20 with 600mm/min" (10mm/s)

philparvin, 2021-11-01:
Thank you for responding, I will experiment with that setting to see if I
can get back to burning at less intensity by letting it burn longer on a
spot.
I would like to see a general settings sheet that different people could
post to that shows their setting for whatever laser spec, and material and
a pic of the result .
Further, an explanation of all the pertinent variables ans what they do.
It seems that everyplace that I look it gives basic settings and a few
things that have worked, BUT there doesn't seem to be anywhere that gives
all the information for ALL the settings and general explanation for all.
Anyway, I also have issues that I don't know what to do about, and really
nowhere to ask except blogs and a hope that someone that understands will
respond.   Ugggg
Sorry for the rant but I think it will take some people interested in real
answers to get together and resolve these issues.
I had a project to burn a name on a handle and it wouldn't stay strait. I
don't even know where to start to resolve, but I don't like putting out
pieces that aren't high quality.

sergejzr, 2021-11-02:
AFAIK there is no single answer to the setting problem. Each laser is a bit different and there are not always standardized materials. LaserGRBL has a built-in material database that can be used to save settings you experimented with.
Generally, thinking of this problem, there should be test files that you can use to automatically produce cuts at different settings. Like this guy did (at 6:40) https://www.youtube.com/watch?v=uXOigFsfIwg&t=400s May be I will try to produce such file in the next months