arkypita/LaserGRBL #1640: X Y coordinates not displayed (always zero)

DanielTheGreat, 2021-11-01:
LaserGRBL's display (top-right of image display area) of X and Y position is always 0, regardless of where I jog the head (or move it with G0...).

**To Reproduce**
Home the engraver head, make this the coordinate origin.
Do a G0 move to some other location.
The XY display always shows 0,0.

**Expected behavior**
Should show the actual head coordinates.

Screenshot attached:
[image]

**Hardware and software configuration**
 - Windows 7 Pro 64-bit
 - VoidMicro LD2429O laser engraver
 - Grbl version 1.1h
 - LaserGRBL version 4.6.0

arkypita, 2021-11-01:
The message "handle broken ok" detects that there is a disturbance on the communication between the machine and your PC.

The positions are established in LaserGRBL on the basis of the messages that the machine periodically sends.

If you don't see any position it means that the machine is not sending them, it is sending them wrong, or they are not recognizable because they are damaged by disturbances.

arkypita, 2021-11-01:
Based on what I see on their website http://www.voidmicro.com/Manual/OMachine-en.html?page_download look very probable that they use a modified version of grbl firmware. Are you sure you switch to the grbl firmware? http://www.voidmicro.com/Manual/Updata.html?page=page_Manual

DanielTheGreat, 2021-11-02:
Mrjavaci and arkypita, thank you both for your comments.

I tried $10=2, but it fixed nothing (I don't understand how it could – it's just changing the reporting details).

It puzzles me that "handle broken ok" signifies a communications issue – **strange message** for that!  It also puzzles me that it occurs **only** after a $H command, never any other time.  Could it be that this lousy VoidMicro machine, with probably 'non-standard' GRBL, as you suggest, is simply sending an unrecognised message after a Home command, which LaserGRBL doesn't understand and thus interprets as a comms error?
[image]

In any event, it looks like I need to return this machine for a refund, as I was about to do before the seller **finally** told me about changing the machine's mode to 'GRBL-compatible'.  Clearly it's not, fully.

Daniel

arkypita, 2021-11-02:
Each time LaserGRBL send a gcode line, the controller reply with "ok".

When those ok's are not recogized because they are blended inside some other message, LaserGRBL detect this situation and write "handle broken ok"

arkypita, 2021-11-02:
Can you copy/upload what is in your logfile?
You can see logfile from LaserGRBL menu ?, Open session log.

You can also record all the communication between LaserGRBL and engraver by activating "communication log" from the same menu

DanielTheGreat, 2021-11-02:
arkypita, thanks for your ongoing support – much appreciated by this 77YO novice!

First of all, my apologies – I must have had a 'brain-snap' when writing my initial post.  My 'office' PC is Win7, but my 'workshop' notebook is running WinXP SP3.  Sorry.

[Temp.zip](https://github.com/arkypita/LaserGRBL/files/7458528/Temp.zip)

The attached sessionlog is quite large, covering a span of time.  The comlog is small, being just enabled, a few instructions issued, then closed.

I hope this helps.

arkypita, 2021-11-02:
You are a  77YO novice able to zip a logfile and upload to github. better then 99.9% of 70s 

Looking at logfile I clearly see the issue. Your engraver is sending status message (the message that tell LaserGRBL the status of the engraver, the position of laser head, and more) in a wrong format.

The error is that the message sent from your engraver is in this format: 

Idle|**WPos:0.000,0.000**|Bf:15,250|FS:0,0|Ov:100,100,100|A:S

While LaserGRBL Expect this format:

Idle|**WPos:0.000,0.000,0.000**|Bf:15,250|FS:0,0|Ov:100,100,100|A:S

You can see that 3 numeric values are expected for the position (X, Y, Z) even if the machine is two-axis.
This is a non-standard implementation of grbl firmware because I cannot find anything that tell that Z is optional.

Here is firmware documentation: https://github.com/gnea/grbl/wiki/Grbl-v1.1-Interface#real-time-status-reports

[image]

DanielTheGreat, 2021-11-03:
Arkypita,

Many thanks for looking into this, confirming your (and my) suspicions.

I have contacted the eBay seller asking them to either provide a controller with standard GRBL, or else take the machine back for a refund.

Your help is very much appreciated.  (PS: I'm a retired electronics engineer, so reasonably conversant with computer technology. :-)

Daniel

arkypita, 2021-11-03:
Ps. It is really easy to me to change LaserGRBL code to handle this non-standard status message. I can release a test version in few time

DanielTheGreat, 2021-11-03:
Wow, that's **great service**, arkypita!  Having little faith in the Chinese seller being able to get the manufacturer to fix the firmware, this would be a wonderful outcome.

Absolutely no hurry, but if you can do that and let me know when ready to test, I will check it with my engraver and provide feedback (and logs if desired) to you.

Many thanks,
Daniel

DanielTheGreat, 2021-11-03:
Arkypita,

If you're going to be revising the LaserGRBL software, perhaps you could look at another (minor) issue I've noticed.  The program comes with a few default 'buttons', such as 'Home'.  But I could not find a way to either delete or edit any of the default buttons.

On the VoidMicro engraver, the homing switches are at the 'top-left' of the frame, whereas I want my job 'home' position to be bottom-left.  I have achieved that with a custom button, but that means I now have two 'Home' buttons cluttering the button area.  It would be nice to be able to add my extra code to the default button instead, or else delete the default Home button and have just my custom one.

As I said, not important, but if it's also easy...

Regards,
Daniel

arkypita, 2021-11-06:
The homing button send "$H" command. There is no way to change it because it is not needed to change.
The homing procedure is totally managed by the controller board.

DanielTheGreat, 2021-11-06:
Yes, I understand what it does.

But it's not enough!  The $H command leaves the head in the 'top'-left corner of the XY frame (where the two home switches are located), whereas the 'normal' XY origin (for any CNC work) it usually in the 'bottom'-left, so X+ moves to the right and Y+ moves 'up'.

So to fully 'home' my engraver, I need to do:
$H
G92X0Y0 (because I don't know what coordinates the controller thinks it's at after $H; I KNOW it's not Y0!)
G0Y-290 (move to bottom-left)
G92X0Y0 (new origin)

I have made a custom button to do all that, but now I have TWO buttons named 'Home', and I'd like to free-up the space for additional custom buttons be removing one of those Home buttons.  Either delete or edit the default Home button (which as it stands, is useless to me, as it doesn't complete the 'homing' manoeuvre for my machine).

Regards,
Daniel

arkypita, 2021-11-06:
Can you give me a feedback on this: https://github.com/arkypita/LaserGRBL/releases/download/v4.6.2/install.exe

DanielTheGreat, 2021-11-07:
Thanks arkypita.  I installed version 4.6.2 and ran a few tests today (Sun).

The XY coordinates now display as expected, but the $H (and a few other commands) still prompt a 'broken handle' error message (see screen-shot).  No big deal; at least I can now see where the head is positioned, and engraving works.

Still can't delete or edit the default Home button, though! :-)

[image]

Regards,
Daniel

161975, 2021-12-11:
DanielTheGreat,
The following is a bit off your subject........
I just read your discussion with Arkypita.
I understand that you run a VoidMicro laser O-model. I have the same but the laser is a LD4070HI (I or 1 ?)
Running the VM on the new laser-GRBL 4.6.2. -->X&Y motors run as expected, BUT The laser  refuses to function what so ever, it won't even run the fan,     nada.

FYI. VM has issued a updated version, but that won't even connect, as it insists to connecting to a non existent comport.
Can you please tell me HOW you got the laser to work on your VM?
many thanks
adrian

DanielTheGreat, 2021-12-13:
Adrian,

Sorry for the late reply – I spent three days battling to get my PC (my main one, not the WinXP one I use for the engraver!) dual-booting with Win7Pro64 and Win10Pro64, and only finally succeeded yesterday (after many crashes and image restores!).

Also sorry that I can't really help you.  After the seller finally sent me a video on how to switch the engraver to GRBL mode, I found that LaserGRBL wouldn't drive it correctly.  When Arkypita looked at my logs he diagnosed an incorrect communication string format from the VoidMicro controller, but he then released a new beta that handled that.  From there, I just did some test burns and everything worked as it should, including my laser head.  I didn't have to make any changes, other than the LaserGRBL settings to reverse the Y-axis.

Incidentally, as part of their effort to sort out the communications problems with their engraver, my seller sent me a replacement controller (newer) that was different from the original one.  When I fitted that, even the new beta LaserGRBL wouldn't talk to it, and nothing worked!  I naturally went back to original controller that came with the engraver.

Did your engraver 'chassis' and laser head come from the same supplier?  If so, put the pressure on them to sort it out or else return the lot for a refund.  If they came from different suppliers, I'm afraid you'll need to seek help from people who understand how all these things work (laser heads and their signals, especially).

Daniel

161975, 2021-12-13:
Hello Daniel,
Many thanks for your reply. I recently contacted VM, by email, Explained what was (not) happening and sent them the log file:    Got a one line reply back with a question as to, "Where I bought the machine."
No answer as such,  not even a suggestion to solve the problem. Not very help-full.

I tried VM in Win7pro & 10pro, (Never tried XP) neither one works, Mind you it does work fine with their original software, but I wanted GRBL control as well.
If I buy an other one someday, I now know what NOT to buy
Adrian

DanielTheGreat, 2021-12-13:
Yes, VoidMicro is definitely a company to be avoided.  I bought my engraver through a (Chinese) third-party on eBay, and despite the severe language problem explaining to them what the issues were, they did at least try to do something (though ultimately unsuccessful, as I previously mentioned).

Daniel