arkypita/LaserGRBL #1644: Laser does not Burn without PWM Support.

codingkopf, 2021-11-03:
hello,
I have the following problem.
I ordered [this](https://de.aliexpress.com/item/4000828904159.html?spm=a2g0s.9042311.0.0.6d9e4c4deQsYLm) board and [this](https://de.aliexpress.com/item/4000028895737.html?spm=a2g0s.9042311.0.0.6d9e4c4deQsYLm) laser and now so far everything is set up. the laser has only a red and black cable, so has no PWM support. 
All the stepper motors move as they should, the laser is properly in focus only it just doesn't engrave anything.... zero...
I have already tried the following commands:

M3 S200.000
M3 S400.000 etc.
all this changes nothing... I don't even get a black dot burned into the wood with the laser....
I am really at a loss...
hopefully someone can help...
I have a 12V power supply connected to the board....

LG

StuartB4, 2021-11-03:
Because it's only a 250 mW power. It's more a laser pointer not for engraving.

codingkopf, 2021-11-03:
Oh okay thanks for the Info! I was a bit confused because all worked great except the laser :/
Can you recommend one wich I can use? (with 2 Pin connector)
Thanks for your help
I would be glad about an answer. :)

StuartB4, 2021-11-03:
If you only want to engrave you only need one of about 1.5 to 2 watts.
If you want to engrave and cut you will need something a bit more powerful, 5 to 10 watts really. 
When you buy one look at the Optical Output Power in the specifications. 

A lot of sellers put the input power which seems high, like 20, 40, 50 watts, but the actual laser output is more like 5 watts.