arkypita/LaserGRBL #1645: LaserGrbl v4.6.0.0  (help)

cruz84, 2021-11-06:
LaserGrbl v4.6.0.0
Grbl v1.1h
Wrapper: UsbSerial
Win 7, Ultimate, 6.1.7601.65536, Service Pack 1 (7)
CLR: C:\Windows\Microsoft.NET\Framework64\v4.0.30319\mscorlib.dll 4.8.4110.0 built by: NET48REL1LAST_B

TypeOf exception  [System.Reflection.TargetInvocationException]
Exception message [Exception has been thrown by the target of an invocation.]
Exception source  [mscorlib], thread [Main Thread]
Exception method  [Void SerializationInvoke(System.IRuntimeMethodInfo, System.Object, System.Runtime.Serialization.SerializationInfo, System.Runtime.Serialization.StreamingContext ByRef)]

   ----------- stack trace -----------
   at System.RuntimeMethodHandle.SerializationInvoke(IRuntimeMethodInfo method, Object target, SerializationInfo info, StreamingContext& context)
   at System.Runtime.Serialization.ObjectManager.CompleteISerializableObject(Object obj, SerializationInfo info, StreamingContext context)
   at System.Runtime.Serialization.ObjectManager.FixupSpecialObject(ObjectHolder holder)
   at System.Runtime.Serialization.ObjectManager.DoFixups()
   at System.Runtime.Serialization.Formatters.Binary.ObjectReader.Deserialize(HeaderHandler handler, __BinaryParser serParser, Boolean fCheck, Boolean isCrossAppDomain, IMethodCallMessage methodCallMessage)
   at System.Runtime.Serialization.Formatters.Binary.BinaryFormatter.Deserialize(Stream serializationStream, HeaderHandler handler, Boolean fCheck, Boolean isCrossAppDomain, IMethodCallMessage methodCallMessage)
   at System.Resources.ResourceReader.DeserializeObject(Int32 typeIndex)
   at System.Resources.ResourceReader.LoadObjectV2(Int32 pos, ResourceTypeCode& typeCode)
   at System.Resources.ResourceReader.ResourceEnumerator.get_Entry()
   at System.Resources.ResourceReader.ResourceEnumerator.get_Current()
   at System.ComponentModel.ComponentResourceManager.FillResources(CultureInfo culture, ResourceSet& resourceSet)
   at System.ComponentModel.ComponentResourceManager.FillResources(CultureInfo culture, ResourceSet& resourceSet)
   at System.ComponentModel.ComponentResourceManager.ApplyResources(Object value, String objectName, CultureInfo culture)
   at LaserGRBL.GrblConfig.InitializeComponent() in E:\LaserGRBL\LaserGRBL\GrblConfig.Designer.cs:line 49
   at LaserGRBL.GrblConfig..ctor(GrblCore core) in E:\LaserGRBL\LaserGRBL\GrblConfig.cs:line 26
   at LaserGRBL.GrblConfig.CreateAndShowDialog(Form parent, GrblCore core) in E:\LaserGRBL\LaserGRBL\GrblConfig.cs:line 72
   at System.Windows.Forms.ToolStripItem.RaiseEvent(Object key, EventArgs e)
   at System.Windows.Forms.ToolStripMenuItem.OnClick(EventArgs e)
   at System.Windows.Forms.ToolStripItem.HandleClick(EventArgs e)
   at System.Windows.Forms.ToolStripItem.HandleMouseUp(MouseEventArgs e)
   at System.Windows.Forms.ToolStrip.OnMouseUp(MouseEventArgs mea)
   at System.Windows.Forms.ToolStripDropDown.OnMouseUp(MouseEventArgs mea)
   at System.Windows.Forms.Control.WmMouseUp(Message& m, MouseButtons button, Int32 clicks)
   at System.Windows.Forms.Control.WndProc(Message& m)
   at System.Windows.Forms.ToolStrip.WndProc(Message& m)
   at System.Windows.Forms.ToolStripDropDown.WndProc(Message& m)
   at System.Windows.Forms.NativeWindow.Callback(IntPtr hWnd, Int32 msg, IntPtr wparam, IntPtr lparam)

Inner exception data
TypeOf exception  [System.ArgumentException]
Exception message [Font 'Courier New' does not support style 'Regular'.]
Exception source  [System.Drawing], thread [Main Thread]
Exception method  [Void CreateNativeFont()]

   ----------- stack trace -----------
   at System.Drawing.Font.CreateNativeFont()
   at System.Drawing.Font.Initialize(FontFamily family, Single emSize, FontStyle style, GraphicsUnit unit, Byte gdiCharSet, Boolean gdiVerticalFont)
   at System.Drawing.Font..ctor(SerializationInfo info, StreamingContext context)

arkypita, 2021-11-06:
> Font 'Courier New' does not support style 'Regular'

Look like you have a corrupted windows font

https://www.google.com/search?client=opera&q=Font+%27Courier+New%27+does+not+support+style+%27Regular%27&sourceid=opera&ie=UTF-8&oe=UTF-8

cruz84, 2021-11-07:
thanks, my problem has been solved, regards

closed 2021-11-07 by cruz84