arkypita/LaserGRBL #1646: LaserGRBL work area border in preview area & zooming/panning?

pmcculey, 2021-11-07:
I've been using LaserGRBL for several weeks now & find it pretty amazing! I do have one thing I was wondering about....

When I load a file, the LaserGRBL preview area zooms from 0,0 to the extents of the data in that file - great. I'd love to have a border in that preview area showing the extents of the work area (0,0 to maximum xy travel) and zoom to that if no data is loaded. If data is loaded, then show the 'cuts' relative to the maximum work area. Is that already possible? Is it possible to zoom in/out/pan on that preview area? If so, I've not seen how to do this so far.

I'm using LaserGRBL with a home built machine. All is good apart from the issue above which would be a real 'nice to have'.

Thanks for any info.

nicolab67, 2022-08-14:
I would also find it very convenient to have a zoom function!

thx

myB0B, 2023-05-14:
yes! please add zoom feature

vincentBenet, 2024-05-14:
Zoom feature may be very usefull!

arkypita, 2024-05-14:
https://youtu.be/P2wsZlC67os