arkypita/LaserGRBL #1647: Custom Variables for Buttons and GCode Head

wiring80, 2021-11-08:
Hey, please excuse me if i'm wrong.

For custom buttons we can set variables like "[bottom]" of the loaded image, and so on.

In LaserGRBL Settings there is the ability to create HEAD GCodes also with that variables, if i understand right - descripts that we can use the same syntax like in custom buttons.

So my thought was, if i use the variables bottom und top for the image margins, so i can use this value to set my offset with G10 for offline engraving, cause i use negative coordinates.

Am i right with my thoughts or im confused ?

Because, and thats the problem - this variables doesnt get any values. If i export the GCode, the variables still exists in the HEAD of the GCode. I didnt try it yet with the buttons, only with the HEAD Gcode.

Many Greets, Sven

arkypita, 2021-11-30:
Yes, you are right.
There is an issue in LaserGRBL export where variables are not converted to values

[image]

closed 2021-12-08 by arkypita

arkypita, 2021-12-08:
Let me know if v4.6.3 (pre-release) fix this issue: https://github.com/arkypita/LaserGRBL/releases/download/v4.6.3/install.exe

commit f88d05e2 (2021-12-08): fix #1647

Add expression evaluator in file export