arkypita/LaserGRBL #1651: SVG wrong positions after Import to LaserGRBL if SVGs are organiced in groups in Inkscape

powelllens, 2021-11-09:
**Describe the bug**
I have been working with lasergrbl again for a long time and after my first attempts in Inkscape I had the problem that the svg graphics were scattered all over the surface after the import. Simple geometries were not affected but text that was converted into paths.

I could not see any pattern and even tried to trim the offsets by active manipulation in Inkscape, which worked but was very unattractive. After I could read some problems with version 0.92 I tried version 1.1.1 and version 0.91 afterwards. With the same file always the same result. I also tried several older LaserGRBL versions, all with the same result.

By chance I found out that the offsets change when I change the part groupings in inkscape. after a few tries I could see that after removing all groupings in inkscape the positions were correct.

As soon as elements are grouped with the function "group" the effect occurs again. It gets worse when you combine groups with groups and set deeper layers - As already written simple geometries remain unaffected.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Inkscape (1.1.1) and add a circle, add a text and convert the text to paths
2. group the text an the circle
3. copy and paste the goup a few times and add it to the site
4. save the svg and import the results in LaserGRBL (4.2.6)

**Expected behavior**
The positions of paths and text should alwasy be correct. Independed of groups in Inkscape

Let me know if you need anything else

alastaira, 2021-11-21:
I think this has the same underlying cause as #1455

arkypita, 2021-11-30:
Thank you for this report and thanks to @svenhb for the code, and thanks @falias4 for the fix.
I am currently very busy but as soon as I can I do a test and compile a new version.

arkypita, 2021-12-08:
Let me know if v4.6.3 (pre-release) fix this issue: https://github.com/arkypita/LaserGRBL/releases/download/v4.6.3/install.exe

closed 2021-12-08 by arkypita