arkypita/LaserGRBL #1655: Prewiev is striped and in a bad hue sat i mean. moore in screenshot

Hum77, 2021-11-14:
Hi,
first of all Thank You for SW, is very intuitive and easy to use.
But I have a problem with photo engraving. As You can see on attached screenshot from normal photo converted to graysteps in a prewiev is this:
[image]

In nearly any setting picture looks same or very similar. Other photos do the same.
It is my fault or is it is a bug?
Petr.

arkypita, 2021-11-14:
It's not your mistake, and it's not a problem. It's just how lasergrbl works.
lasergrbl draws the preview based on the generated gcode. when the image is very dense the lines are many, so the image appears darker and artifacts may appear due to the superimposition (on video) of the lines that will not actually exist on the finished work.

Hum77, 2021-11-15:
Hi, thank You for explanation, is absolutely correct, by my mistake i leave (and forgot) 50 lines per mm. 
Problem solved. Thank You for help, fast response and resolving my problem,.

closed 2021-11-15 by Hum77