arkypita/LaserGRBL #1659: Crop image button

Stewestrong, 2021-11-26:
**Describe the bug**
Crop image button turns yellow but does not work. It does not crop the finished engraved image. It constantly goes through the entire engraving instead of the outer outline

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
Kép vágása gomb bekapcsolásával megcsinálja a gravírozást majd a külső körvonal mentén kivágja. Erre számítottam.
**Screenshots**
If applicable, add screenshots to help explain your problem.

**Hardware and software configuration (please complete the following information):**
 - Windows version  Windows 10
 - Engraver brand and model ATOMSTACK New A5 30W Laser Engraving
 - Grbl version (i.e. 0.9j, 1.1f) CH240
 - LaserGRBL version  v 4.6.2

**Additional context**
Add any other context about the problem here.

Stewestrong, 2021-11-26:
v1.1f original

closed 2021-11-26 by Stewestrong

arkypita, 2021-11-29:
One you select the crop tool you should click and drag on the preview to select the area to be crop.

[image]