arkypita/LaserGRBL #1668: System locked when jumping to another image

jrjpontious, 2021-12-05:
Here was my bug report:

LaserGrbl v4.6.2.0
Grbl v1.1f
Wrapper: UsbSerial
Win 10, Professional, 10.0.19042.0 (7)
CLR: C:\Windows\Microsoft.NET\Framework64\v4.0.30319\mscorlib.dll 4.8.4420.0 built by: NET48REL1LAST_C

TypeOf exception  [System.OverflowException]
Exception message [Value was either too large or too small for an Int32.]
Exception source  [mscorlib], thread [Main Thread]
Exception method  [Int32 ToInt32(System.Decimal)]

   ----------- stack trace -----------
   at System.Decimal.ToInt32(Decimal d)
   at LaserGRBL.RasterConverter.ConvertSizeAndOptionForm.AssignMinMaxLimit() in E:\LaserGRBL\LaserGRBL\RasterConverter\ConvertSizeAndOptionForm.cs:line 78
   at LaserGRBL.RasterConverter.ConvertSizeAndOptionForm..ctor(GrblCore core) in E:\LaserGRBL\LaserGRBL\RasterConverter\ConvertSizeAndOptionForm.cs:line 49
   at LaserGRBL.RasterConverter.RasterToLaserForm.BtnCreateClick(Object sender, EventArgs e) in E:\LaserGRBL\LaserGRBL\RasterConverter\RasterToLaserForm.cs:line 175
   at System.Windows.Forms.Control.OnClick(EventArgs e)
   at System.Windows.Forms.Button.OnClick(EventArgs e)
   at System.Windows.Forms.Button.OnMouseUp(MouseEventArgs mevent)
   at System.Windows.Forms.Control.WmMouseUp(Message& m, MouseButtons button, Int32 clicks)
   at System.Windows.Forms.Control.WndProc(Message& m)
   at System.Windows.Forms.ButtonBase.WndProc(Message& m)
   at System.Windows.Forms.Button.WndProc(Message& m)
   at System.Windows.Forms.NativeWindow.Callback(IntPtr hWnd, Int32 msg, IntPtr wparam, IntPtr lparam)

jrjpontious, 2021-12-06:
I need to finish a job VERY SOON as I am going on VC... I can't even continue a new job.  Is there a way to override the Exception Manager to continue?

arkypita, 2021-12-07:
Hi, look like that you have a damaged grbl configuration (the configuration inside your controller).
LaserGRBL in function AssignMinMaxLimit() read parameter from controller and initialize UI inputbox max and min.

Look like you have a weird value maybe in $30 or other conf.

jrjpontious, 2021-12-07:
Thank you for your response. I'll check $30 values.. but it won't be until about a week as I am going on VC. I'll let you know if that helps and let you know..