arkypita/LaserGRBL #1684: Filling Quality threshold when importing raster image is too low

lepasq, 2021-12-26:
When importing a raster image and setting the conversion tool to `Vectorize!`, there is a threshold on the `Filling Quality` of `50.000 Lines/mm`. This threshold is too low for my personal use, and I'd like to use a higher value.  

What is the reason that the threshold is set to 50? Is there a way to increase that limit, or would you consider increasing that limit in an upcoming release?

[image]

arkypita, 2021-12-27:
Ciao Leonardo

Filling Quality is just a number, I would have no problem increasing the limit. Keep in mind that I have already increased this from 20 to 50 at the request of a person who made micro-engravings on jewelry with fiber lasers. 50 means 1200 dpi which is a very high value! You should have a laser capable of focusing <0.02mm spot, in order to take advantage of this resolution.

Keep in mind that that value greatly increases the resources needed (RAM, CPU) if you do large jobs, especially with complex fills like hilbert. I notice that you are using LaserGRBL to make very small PCB masks, so it's probably not a problem for you, but it may be a problem for other users.

Furthermore, going through a vectorization process for such high-precision work is perhaps not the best solution.
Can you tell me more about your process and your tools? Maybe I can suggest a better approach.

Diego

lepasq, 2021-12-27:
Ciao, thanks for your answer! 
I'm using the program for microlithography.
My input files are not PCBs, but micro masks for a chip.

The laser has a spot that can be adjusted between 5 and 10 micrometres.
Using the focus, I can make it a bit larger, but I'd prefer to have a resolution of 10-micrometre resolution or smaller.

arkypita, 2021-12-27:
LaserGRBL is designed for applications where jobs size are usually > 1cm, with details of the order of millimeter. Consequently interface and code works on these assumptions. For example the generated g-code is rounded to 2 decimal places, and it is more than enough.

I think that a trick could be used to work with very small projects and with sub-millimeter details.

I think you know that the $100 and $101 parameters indicate how many motor steps are required to move the laser 1mm. When this data is configured wrong, the image is created with the wrong size. For example if the machine physically requires 80 motor steps to make a millimeter but the $ 100 parameter is set to 160, when you send a 1mm movement command 160 motor pulses are generated and then the laser moves 2mm (instead of 1mm).

My proposal is to divide the $100/$101 values by 10, this way the designs will come to you 1/10 the size.
You will compensate this reduction by giving 10 times the size to your job.

In practice it is as if the base unit in LaserGRBL became the tenth of a millimeter.
You'll probably have to scale the speed, acceleration, and frame size proportionally too, but then it should all work out.

Non ho mai provato, ma potrebbe funzionare! Fammi sapere.

lepasq, 2021-12-29:
That sounds pretty smart actually, I think I'll go with that for now. Thanks a lot for your help and your detailed explanations!