arkypita/LaserGRBL #1723: Laser head unexpectedly stops, shrinks and shifts the engraved image

Gamgrant, 2022-01-16:
**Describe the bug**
A laser head stops in the middle of the engraving while still emitting a laser beam. Once it resumes the movement, it doesn't reach the edge of the photo. When the laser stops, the engraving line isn't shifted by the fixed offset, it varies from line to a line. Typically, if the engraver stops at a given line, then both white space and the engraving portion of the line shrink (see the photo below). Usually I do not have those issues when I start engraving, but in a minute or 2 they appear (laser doesn't stop at the same point either, so I do not believe there is an issue with the code)

**What I have tried (none of it helped):**
 - checked the tension of the pulley belts - tension isn't a problem
 - bought new short usb cable with ferrite chokes, which could mute the electrical noise from the stepper motors (if any)
- lowered the velocity to 1600 mm/min and acceleration down to 20 mm/s^2 to make sure that the laser head can catch up with the lasergrbl commands
- checked the code with $C command to see the code is corrupted (no errors were found)
- changed the streaming mode from buffered to synchronous and to repeatOnError, but that didn't help
- enabled "Unidirectional engraving" to ensure that the laser will engrave only from one side of the image, but due to the unexpected stops the engraving would start from the different points

**To Reproduce**
Steps to reproduce the behavior:
start engraving

**Expected behavior**
no unexpected stops are expected to happen 
**Screenshots**
[image]
[image]
[image]
[image]

https://user-images.githubusercontent.com/94784561/149670024-7423de84-13fa-4e3c-ad34-171ec3979d2f.mp4

**Hardware and software configuration (please complete the following information):**

 - Windows version: Windows 10 Home
 - Engraver brand and model:  OXLasers 3.5W 450nm; Manufacturer : zhuo long dian zi; ASIN:B089LRSL1Z
 - Grbl version (i.e. 0.9j, 1.1f) 1.1 h
 - LaserGRBL version v.4.6.2

**Additional context**

arkypita, 2022-01-16:
Could you upload your session log? LaserGRBL, menu "?", open session log.

Gamgrant, 2022-01-16:
[sessionlog.txt](https://github.com/arkypita/LaserGRBL/files/7877731/sessionlog.txt)
Here it is! Thank you for your prompt response!

arkypita, 2022-01-17:
Log and video doesn't give me any hint.
How is your $32 value?

[image]

arkypita, 2022-01-17:
please post the whole $$ configuration

Gamgrant, 2022-01-17:
My $32 is 1. Here is  a screenshot of my configuration:
[image]

brakthehun, 2022-08-17:
Looks like your drivers are overheating.

Gamgrant, 2022-09-02:
@brakthehun, after adjusting the voltage on the Stepper Motor Driver Module, the issue got resolved. Thank you!

closed 2022-09-02 by Gamgrant