arkypita/LaserGRBL #1724: Telegram Bot not working

HomineLudens, 2022-01-16:
Hi, anyone facing issues registering to laser grbl bot?
I've try for a long time to register but every time I've this error

Cannot retrieve your code now, please retry later. [2]

[image]

arkypita, 2022-01-16:
Ciao

Don't know if someone can answer to you, except me. This feature is used only by 26 people, don't know if other people try using but have issues, or what.

I have added some logging to the code, if you try to re-do the register procedure I can tell you more.

HomineLudens, 2022-01-17:
Ciao!

The feature is really interesting, maybe still not know by many people.

I've try again a couple of minutes ago.
Let me know if I can make any other test.

arkypita, 2022-01-17:
Found some issue in handling null values, i think it's fixed now.

HomineLudens, 2022-01-17:
Awesome! Thanks for the super quick fix.

closed 2022-01-17 by HomineLudens