arkypita/LaserGRBL #1732: Appcrash after loading 4.7.1.0

TherealRCranium, 2022-01-30:
**Describe the bug**
App can flash 1.1 but when trying to connect the app crashes. Previous version was loaded and worked fine.
**To Reproduce**
Steps to reproduce the behavior:
1. Go to '..Open Laser GRBL Com and baud show correct. Using Nano board based burnewr .'
2. Click on '..Connect Icon..'
3. Program hangs for a few seconds then crashes.
4. Problem Event Name:	APPCRASH
  Application Name:	LaserGRBL.exe
  Application Version:	4.7.1.0
  Application Timestamp:	61d4b2ea
  Fault Module Name:	clr.dll
  Fault Module Version:	4.8.4110.0
  Fault Module Timestamp:	5de6db95
  Exception Code:	80131506
  Exception Offset:	0000000000492b76
  OS Version:	6.1.7601.2.1.0.256.48
  Locale ID:	1033
  Additional Information 1:	e982
  Additional Information 2:	e98281cd699af7eee8bd21bae7a0476f
  Additional Information 3:	4566
  Additional Information 4:	4566fcd1449982b2ae6ba37619c5a0cd

**Expected behavior**
Previous version had no issues. controlled everything fine.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Hardware and software configuration (please complete the following information):**
 - Windows version Win 7 Pro 64 bit on Dell prcision laptop
 - Engraver brand and model
 - Grbl version1.1
  - LaserGRBL 4.7.1.0

**Additional context**
Going to try and find the installer I used for the previous version to see if reverting works.

TherealRCranium, 2022-01-30:
Uninstalled and rebooted, Installed 4.6.0 and everything is working again.

closed 2022-01-30 by TherealRCranium

arkypita, 2022-01-30:
Don't know what could be the cause. v4.7.1 is used by 34% of LaserGRBL users without any issue, and nothing is changed between 4.6 and 4.7 that can cause this kind of problem.
I think it was something in your specific PC. I suggest to reinstall or update .NET Framework component of your PC and try again with v4.7.1 (if you want).

TherealRCranium, 2022-02-01:
Thanks. I looked over the update and didn't see anything that was needed so
since 4.6 is stable I'll  stay with that.