arkypita/LaserGRBL #1733: SVG importing wrong on Y-axis

Sixleggedbuffalo, 2022-02-01:
I'm importing a SVG and the vectors are appearing at the wrong location on the y-axis. Not just mirrored into the negative but completely the wrong location. I've completed a few projects using LaserGRBL and never had an issue like this. All my other files work fine still. Opening the SVG in Inkscape (where I created it) or Edge shows it at the correct location.

Steps to reproduce the behavior (running in emulator mode):
1. Import SVG
2. Use M3 mode
3. S-max at 1000

**Expected behavior**
Vectors should start from 0,0 coordinates.

**Screenshots**
Attached.

**Hardware and software configuration (please complete the following information):**
 - Windows version 10
 - LaserGRBL version 4.6.2
[image]

arkypita, 2022-02-01:
can you share the SVG?

Sixleggedbuffalo, 2022-02-01:
Many thanks for your quick response! See attached
[image]
.

Sixleggedbuffalo, 2022-02-01:
Perhaps an enhancement suggestion could be to implement a way to define the origin coordinate for an imported file.

Sixleggedbuffalo, 2022-02-01:
My apologies I should have looked harder in previous issues.

The fix here https://github.com/arkypita/LaserGRBL/issues/1126 applies to my issue too.

Many thanks for your time - kind regards