arkypita/LaserGRBL #1742: Sculpfun S9

MaulGB, 2022-02-07:
Hi,

how can i add laser heads Sculpfun S9 to the program Laser grbl? 

Best regards
Grzegorz

gmmanonymus111, 2022-02-07:
You can't 'add a laser head' to LaserGRBL. If you meant connecting your engraver, please refer to [LaserGRBL's manual](https://lasergrbl.com/usage/arduino-connection/).
Since it is not a LaserGRBL issue, next time please open a discussion instead for such questions.

arkypita, 2022-02-08:
If you mean add to "Material definition DB" you can edit this DB and add everything you want.
Menu "Grbl" > "Material DB".

All that entry are values provided by manufacturers, which I add when they send me this data. I am waiting for Sculpfun to send me the values for S9, however you can use those of any other compatible power machine, as they are only reference values that you will have to adapt a little according to your tests on the different materials.

closed 2022-02-08 by arkypita