arkypita/LaserGRBL #1747: GRBL 1.1f upload to laser engraver - wont move x and y axis

sccottrill, 2022-02-09:
I successfully flashed my laser engraver with laser grbl to 1.1f on a machine I just put together.  It is a 6550 laser engraver could be an eleksmaker model.  I purchased it a while ago, have no original documentation, have installed the CH23* driver and would like to start playing with it for a project I am working on. 

The problem: I cannot jog neither the x nor y axis.  I even tried bringing a picture in and sending it as a sample job to the machine - nothing.

I did hours of searching as well as modified the configuration file per settings from youtube videos describing the process with no success.  I am hoping some of you may be able to help me.

[image]

arkypita, 2022-02-09:
Your grbl config has noting wrong at first look.
If motor doesn't move at all I guess an hardware issue.

alastaira, 2022-02-19:
What happens when you *try* to jog, or run a file? Can you post a screenshot of the LaserGRBL console log here? If the commands appear to be being received and processed correctly (i.e. the controller is acknowledging them with an "Ok" message), then it suggests a mechanical/electrical fault with the way you've wired your motors.

sccottrill, 2022-02-20:
The problem has been solved.  Thank you everyone who responded.  

Since I had gotten this device without a manual and only had what I could find online I'm surprised there isn't more mentioned about the operation of the device even in the YouTube videos and such.  I had mistaken the switch on the board for something laser related as is mentioned in some videos of slightly different machines instead of an on off switch.  I also wasn't are that plugging the cable into the computer allowed for the lights to come on and being able to flash the board firmware without actually having to turn the board on and off.  I feel a little silly for overthinking the problem and the time I spent pursuing other issues but these devices are new to me.  I'm use to on off switches completely controlling power to all devices instead of this just controlling everything down stream of the chip.

I am currently playing around with settings and focal distance with test prints to test the capabilities and settings of the machine.