arkypita/LaserGRBL #1759: Automatic cooling sometimes works and sometimes doesn't

Dranim, 2022-02-13:
Hi!. Having all the requirements for this option to work, I activate the option and put save.

Sometimes by doing this the automatic cooling works, but other times it doesn't, even though i do exactly the same thing (activate the option and put save).

Has this happened to anyone else, any idea why this is happening or how to fix it?.

arkypita, 2022-02-15:
Hi @Dranim 
I checked the code to refresh my memory as to how it should work.

AutoCooling it is conditioned by the fact that the machine actively supports this functionality, that is, that it has grbl 1.1 and is configured to have the laser mode active ($32 = 1). Please verify both this condition.

arkypita, 2022-02-15:
If you have more than one engraver, it could be that switching from one to the other LaserGRBL does not re-read the configuration and therefore keeps those of the previous machine (maybe the one without $32=1).
In this case, have the values read again by opening the grbl configuration window.

Dranim, 2022-02-17:
I found the solution to this by accident, it is something strange to solve.

For the automatic cooling to work I have to start the job, then pause it. 

Enter the automatic cooling and uncheck it and mark it, while the job continues to pause and put save.

And then resume work, in this way the automatic cooling worked every time.

arkypita, 2022-02-17:
Hi @Dranim 
There may be some correlation because auto cooling and pause/resume are related (they both use the same `feed hold` and `resume` [commands](https://github.com/gnea/grbl/wiki/Grbl-v1.1-Commands#ascii-realtime-command-descriptions)).

Exactly for this reason auto cooling must be inhibited from acting when the user manually presses the feed hold button (in particular, it must **absolutely** not restart the work if it has been suspended by the user) and should re-start its work when the user resume the job. Also if auto-cooling has suspended work, but the user presses the resume button, the user action must restart the work early.

Relying on the [source code](https://github.com/arkypita/LaserGRBL/blob/5d06b113ef8a6f3e3d893c6ea8ff8b4601f5cc5f/LaserGRBL/Core/GrblCore.cs#L2365) here is how autocooling work:

[image]

- `ManageCoolingCycles()` it is called continuously
- `AutoCooling` means "AutoCooling is enabled && Grbl >= 1.1 && $32=1"
- `InProgram` means "engraving started and not completed jet"
- `!mHoldByUserRequest` means "job suspend not requested by user"
- `NowCooling = [...] ` discretizes the true/false and false/true variation of the time condition and commands the stop and start

According to this "`AutoCooling && InProgram && !mHoldByUserRequest`" test, if the "mHoldByUserRequest" variable becomes misaligned (ie it stays high when it shouldn't) the autocooling may stops doing its job.

`mHoldByUserRequest` is settled "true" when you press the feedhold button OR if you have a custom button that send the "!" command.

`mHoldByUserRequest` is settled "false" only when you manually press the "Resume" button

So there may be some cases where this stays high... for example if you press the feedhold button and then reset (lightning bolt) and/or maybe other cases.

Are you sure this is not the case for you?

Dranim, 2022-02-17:
I'm not that into this and I don't understand some of this stuff. 

For me if the automatic cooling option is activated it should always work, no matter if the program is restarted or something else.
 
Like other options in the configuration bar that if they are activated or deactivated there they remain that way, no matter the reboots or changes or whatever.

They just stay as they are left. 

But hey, I'm not as into the subject as I said and for me, with the solution I found, it works for me. 

Thanks for the answer. Greetings.

jomo487, 2022-02-19:
I have noticed an issue which might be related. Automatic cooling stopped working when I do multiple passes that the first pass doesnt exceed my 10 minute limit, but multiple passes do and the laser stays on the whole time.

arkypita, 2022-03-12:
> I have noticed an issue which might be related. Automatic cooling stopped working when I do multiple passes that the first pass doesnt exceed my 10 minute limit, but multiple passes do and the laser stays on the whole time.

You are right.

Autocooling cycles is computed on "ProgramTime" that is the running time of the current program.
Since multiple cycles are simply the same program repeated multiple times the "ProgramTime" reset at each cycle, so autocooling never trigger.

I don't have a global clock for multiple pass, so it is not easy to fix this issue, but i'll think about it.

arkypita, 2022-03-29:
Most of these problems was fixed in version 4.8

closed 2022-03-29 by arkypita

MinraVex, 2025-06-19:
so if i go back to version 4.8 my auto cooling will work?   because its not working and i damn need it