arkypita/LaserGRBL #1766: "Ooops! Somethings went wrong"

EngraverGlassman, 2022-02-19:
Hello everybody, i have trouble, the application behaves strangely. After importing the image, code is created,

[image]

 and sent to Arduino Uno and the device works, but**nothing appears** on the preview window (see photo atachment)

[image]
.
Also, when I want to configure GRBL via the menu , an error message pops up: "Ooops! Somethings went wrong"

[image]

 (similarly even when I press the "Hot key" button).

[image]

 I can't get into the GRBL settings at all, as you can see from the attachment. So I can't set anything at all ..
My computer's configuration can be seen in the error message in the attachment. 
Does anyone know the answer, please? I don't know what this is about. 
Thank you much.

arkypita, 2022-02-19:
This is an error in a system library that cannot be fixed by me. Could you try to update Microsoft .NET Framework?

arkypita, 2022-02-19:
https://dotnet.microsoft.com/en-us/download/dotnet-framework

EngraverGlassman, 2022-02-19:
Ciao Arkypita, grazie per la tua risposta. Neanche io penso che sia un bug di LaserGRBL. Ho aggiornato il programma stesso e ho anche aggiornato il Framework all'ultima versione per un altro programma (e probabilmente non ho avuto problemi da allora). Comunque, grazie, ho provato a vedere se qualcuno ha avuto lo stesso problema e ho subito saputo la soluzione. Ho notato che AntononGalin probabilmente ha lo stesso problema qui: https://github.com/arkypita/LaserGRBL/issues/1729.
Altrimenti sto sperimentando con il Laser GRBL per controllare una macchina per incisione DIY Impact per immagini di granito, ho sostituito la testina laser con un percussore diamantato elettromagnetico che gestisce la S ...
Mi scuso per la traduzione italiana tramite google :) ..

arkypita, 2022-02-19:
Thanks for the feedback and the translation (is perfect)

rtramell, 2022-03-21:
I too am having problems with the Opps window, problems connecting with the com port and stating that it having problems with the engrave(3d) path after the update: so frustrated

EngraverGlassman, 2022-03-21:
> I too am having problems with the Opps window, problems connecting with the com port and stating that it having problems with the engrave(3d) path after the update: so frustrated

I tried the Framework updates, but only reinstalling the windows solved my problems ...