arkypita/LaserGRBL #1767: Add more scanning resolution!

VICLER, 2022-02-19:
Hello! I need more resolution than a 0.05mm (20 lines/mm). I'm trying to make some small pcb boards using LaserGRBL, but the maximum resolution is just not enough for this purpose.. I have tried to use another software where I can set the resolutiion to <0.05 (actually 0.02mm) and it looks much more better on the pcb (much more details). 

**Describe the solution you'd like**
please change the max resolution to 50 lines/mm

And on the resolution helper window explains, that the resolution should not exceed max full engine steps, but even with a 0.9deg steppers there is a limit of 20 lines/mm (the same as for 1.8deg motor)

StuartB4, 2022-02-19:
[image]

closed 2022-02-19 by VICLER

arkypita, 2022-02-19:
This may help too: https://github.com/arkypita/LaserGRBL/issues/1684#issuecomment-1001660398