arkypita/LaserGRBL #1783: Laser GRBL not opening in my computer

tonyjavi, 2022-03-07:
Hey,

I have been using laser grbl with my small laser engraver for 2 years now and suddenly the application stopped opening in from my desktop so I can no longer use it.

I tried updating and re-installing but nothing seems to work. Neither there are any information banners popping up.

Any ideas?

Hardware and software config.
 - Windows version 10
 - Engraver brand and model. TOPQSC CNC 2 axis, 65x50 cm
 - LaserGRBL version 4.7.0

Thank you all!

tonyjavi, 2022-03-07:
Hi, 

I looked thoroughly for this folder in my user but did not find it. Do you think of any other possible solution?

Thanks again,

StuartB4, 2022-03-07:
Is the AppData folder showing, it can be a hidden folder.
If you click C, then Users, then "your name" folder, then the View menu and Options on the right you get the option to show or hide certain files and folders. Click Options and then the View tab. 

[image]

[image]

tonyjavi, 2022-03-07:
Thanks for the tip! Now I can visualize the file but can not open it as it's a .bin folder.

I tried to open with PoweISO app but can't. Do you recommend me other app or way to open and clean this folder? Or should I just delete it?

Thanks,

[image]
[image]

StuartB4, 2022-03-07:
If you are going to reinstall LaserGRBL you can delete any other LaserGRBL folders first.

tonyjavi, 2022-03-08:
I deleted all of them and any other file from my laptop and reinstalled but same thing happens. I download it and create a desktop shortcut but the program does not open. 

Now when I search in User>AppDate some folders are missing.

[image]

StuartB4, 2022-03-08:
I don't have any other folders in there, just the files that Lasergrbl creates. May be they are not created because it does'nt start properly. 

[image]

StuartB4, 2022-03-08:
You say you install it and create a shortcut to the desktop, but lasergrbl should do that. 
Are you using the correct shortcut.

arkypita, 2022-03-08:
@tonyjavi can you also look inside windows event viewer (Application and System log) if you have any record about LaserGRBL or .NET framework near the clock time when you open LaserGRBL?

https://www.howtogeek.com/123646/htg-explains-what-the-windows-event-viewer-is-and-how-you-can-use-it/

tonyjavi, 2022-03-10:
Hi all, 

Luckly I upgraded my windows to version 11 and all of sudden the program started working again.

Thank you for the tips,

closed 2022-03-10 by tonyjavi