arkypita/LaserGRBL #1786: Upgraded to latest version; image now not the same size?

maillemaker, 2022-03-08:
**Describe the bug**
I upgraded to the latest version of LaserGrble today.  

Now when I open my image (13" wide x 10" tall) and tell it to find the center, it is not in the place it used to be (it is off by a few inches).  If I manually then drag the laser to my old center and then tell it to go home, it crashes the motors as it goes -X/-Y past zero.

**To Reproduce**
Steps to reproduce the behavior:
1. Open .jpg file.
2. Do Vectorize with 6mm diagonal fill, B&W.
3. On size screen 1500 fill 1200 border speeds.  Height 254 mm (10").  Width auto-populates correctly.
4. press Center button.  Laser tracks to where it thinks center is, which is not where it used to be.

I uninstalled the latest version of lasergrbl and installed the original version and it works fine again.  

**Expected behavior**
Center should be found where it was on previous version.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Hardware and software configuration (please complete the following information):**
 - Windows version - 10
 - Engraver brand and model - Eleksmaker 3.0 Pro.  Board is Manna SE 3.2
 - Grbl version (i.e. 0.9j, 1.1f) - unknown
 - LaserGRBL version Was running 3.0.14 worked fine.  Upgraded to 4.7.1 saw problem. Went back to 3.0.4 works fine.

maillemaker, 2022-03-09:
I updated my Grbl to 1.1h, and re-updated from 3.0.4 to 4.7.1.  Problem has gone away.

StuartB4, 2022-03-09:
Thats good. 👍