arkypita/LaserGRBL #1799: TinyUSB connect LaserGRBL v4.8.0 failure

CYCHEN97, 2022-04-01:
Use LaserGRBL v4.8.0

My board uses TinyUSB of ESP32S2, without CH340 serial chip, to connect LaserGRBL. When I click the connection, it shows Connecting in the lower right corner of the page, but all the buttons are gray, so I can't click. Then, after dozens of seconds, The Connecting link in the lower right corner becomes Disconnected.
[image]
[image]

When the machine is connected to the computer, press the button on the machine to restart the machine. Sometimes the connection will be successful, but most of the time the connection will fail.
[image]

How can this be solved?

Mrjavaci, 2022-04-01:
Click the "?"  button.
Click the "Active extended log" button.
Write where to save the file "comlog.txt"'.
Try to connect to the machine again, then dump the contents of that file here.

Let us help you understand the error.

arkypita, 2022-04-01:
Hi

When connecting to a Grbl compatible device LaserGRBL expect to receive back some messages (that is not shown inside console) that tell LaserGRBL the current machine position and state. This is called the "[realtime status message](https://github.com/gnea/grbl/wiki/Grbl-v1.1-Interface#real-time-status-reports)". If not LaserGRBL cannot consider itself connected to a valid Grbl device.

Using comlog as suggested by @Mrjavaci you can see what's inside the communication and so what's wrong/missing. Also SessionLog could be useful for diagnostic.

Here is how a valid comlog look like: [comlog.txt](https://github.com/arkypita/LaserGRBL/files/2636346/comlog.txt)

arkypita, 2022-04-01:
Please also check this thread: https://github.com/arkypita/LaserGRBL/issues/795

CYCHEN97, 2022-04-01:
I connected twice but failed to connect properly. Then I restarted my machine by pressing the button and clicked connect again. Below is the log information
[comlog.txt](https://github.com/arkypita/LaserGRBL/files/8395409/comlog.txt)

arkypita, 2022-04-01:
Could you share the sessionlog too?

arkypita, 2022-04-01:
The first two session LaserGRBL open the serial port but the machine is silent, no line sent from the machine (no "rx" lines) so, after a timeout of 10 seconds LaserGRBL consider itself not connected to a machine.

```
00000	00012033	log	Recording session started @ 2022/4/1 14:58:13
00001	00013670	com	Close COM8 [CORE]
00002	00017819	com	Open COM8 @ 115200 baud  (UsbSerial)
00003	00017828	tx	[3F]
00004	00019837	tx	[3F]
00005	00021837	tx	[3F]
00006	00023842	tx	[3F]
00007	00025829	tx	[3F]
00008	00027832	com	Close COM8 [CORE]
```

On the third try the machine actively reply, and so LaserGRBL is fine in considering the macine connected, as you can see there is a tx-rx dialog between LaserGRBL and the machine.
```

00016	00146177	com	Open COM8 @ 115200 baud  (UsbSerial)
00017	00146192	tx	[3F]
00018	00146196	rx	<Idle|MPos:-30.000,-30.000,0.000|FS:0,0|Pn:XY|WCO:0.000,0.000,0.000>\r\r
00019	00146231	tx	$$\n
00020	00146233	rx	$8=50\r\r
00021	00146239	rx	$7=50\r\r
00022	00146241	rx	$0=10\r\r
00023	00146242	rx	$1=255\r\r
00024	00146244	rx	$2=0\r\r
00025	00146245	rx	$3=0\r\r
00026	00146247	rx	$4=0\r\r
00027	00146248	rx	$5=0\r\r
00028	00146249	rx	$6=0\r\r
00029	00146251	rx	$10=1\r\r
00030	00146252	rx	$11=0.010\r\r
00031	00146254	rx	$12=0.002\r\r
00032	00146256	rx	$13=0\r\r
00033	00146257	rx	$20=0\r\r
00034	00146259	rx	$21=0\r\r
00035	00146261	rx	$22=1\r\r
00036	00146263	rx	$23=0\r\r
00037	00146265	rx	$24=3000.000\r\r
00038	00146267	rx	$25=3000.000\r\r
00039	00146269	rx	$26=20.000\r\r
00040	00146271	rx	$27=30.000\r\r
00041	00146273	rx	$30=1000.000\r\r
00042	00146275	rx	$31=0.000\r\r
00043	00146277	rx	$32=1\r\r
00044	00146278	rx	$N1=******\r\r
00045	00146279	rx	$N0=******\r\r
00046	00146280	rx	$100=7000.000\r\r
00047	00146282	rx	$101=7000.000\r\r
00048	00146283	rx	$102=7000.000\r\r
00049	00146284	rx	$100=80.000\r\r
00050	00146285	rx	$101=80.000\r\r
00051	00146287	rx	$102=160.000\r\r
00052	00146288	rx	$110=12000.000\r\r
00053	00146290	rx	$111=12000.000\r\r
00054	00146292	rx	$112=12000.000\r\r
00055	00146293	rx	$120=500.000\r\r
00056	00146294	rx	$121=500.000\r\r
00057	00146296	rx	$122=500.000\r\r
00058	00146298	rx	$130=600.000\r\r
00059	00146299	rx	$131=600.000\r\r
00060	00146301	rx	$132=600.000\r\r
00061	00146302	rx	ok\r\r
00062	00148195	tx	[3F]
00063	00148198	rx	<Idle|MPos:-30.000,-30.000,0.000|FS:0,0|Pn:XY|Ov:100,100,100>\r\r
00064	00150194	tx	[3F]
00065	00150196	rx	<Idle|MPos:-30.000,-30.000,0.000|FS:0,0|Pn:XY>\r\r
00066	00152199	tx	[3F]
00067	00152202	rx	<Idle|MPos:-30.000,-30.000,0.000|FS:0,0|Pn:XY>\r\r
00068	00154208	tx	[3F]
00069	00154211	rx	<Idle|MPos:-30.000,-30.000,0.000|FS:0,0|Pn:XY>\r\r
00070	00156200	tx	[3F]
00071	00156202	rx	<Idle|MPos:-30.000,-30.000,0.000|FS:0,0|Pn:XY>\r\r
00072	00158204	tx	[3F]
00073	00158207	rx	<Idle|MPos:-30.000,-30.000,0.000|FS:0,0|Pn:XY>\r\r
00074	00160204	tx	[3F]
00075	00160206	rx	<Idle|MPos:-30.000,-30.000,0.000|FS:0,0|Pn:XY>\r\r
00076	00162207	tx	[3F]
00077	00162210	rx	<Idle|MPos:-30.000,-30.000,0.000|FS:0,0|Pn:XY>\r\r
00078	00164208	tx	[3F]
00079	00164210	rx	<Idle|MPos:-30.000,-30.000,0.000|FS:0,0|Pn:XY>\r\r
00080	00166193	tx	[3F]
00081	00166197	rx	<Idle|MPos:-30.000,-30.000,0.000|FS:0,0|Pn:XY|WCO:0.000,0.000,0.000>\r\r
00082	00168205	tx	[3F]
00083	00168208	rx	<Idle|MPos:-30.000,-30.000,0.000|FS:0,0|Pn:XY|Ov:100,100,100>\r\r
00084	00170205	tx	[3F]
00085	00170208	rx	<Idle|MPos:-30.000,-30.000,0.000|FS:0,0|Pn:XY>\r\r
00086	00172196	tx	[3F]
00087	00172199	rx	<Idle|MPos:-30.000,-30.000,0.000|FS:0,0|Pn:XY>\r\r
00088	00174196	tx	[3F]
00089	00174199	rx	<Idle|MPos:-30.000,-30.000,0.000|FS:0,0|Pn:XY>\r\r
00090	00176200	tx	[3F]
00091	00176204	rx	<Idle|MPos:-30.000,-30.000,0.000|FS:0,0|Pn:XY>\r\r
00092	00178207	tx	[3F]
00093	00178210	rx	<Idle|MPos:-30.000,-30.000,0.000|FS:0,0|Pn:XY>\r\r
00094	00180203	tx	[3F]
00095	00180206	rx	<Idle|MPos:-30.000,-30.000,0.000|FS:0,0|Pn:XY>\r\r

```

arkypita, 2022-04-01:
I suggest you to turn on Hard-Reset and Soft-Reset on connect, so LaserGRBL sent reset command to the board when connecting, this may help you with this issue.

[image]

CYCHEN97, 2022-04-01:
If I choose soft reset, the machine will restart, and if I choose hard reset, the machine will not restart, and the button can also be enabled normally. I will find more laptops to test, thank you.

CYCHEN97, 2022-04-01:
@arkypita 
I took several laptops for testing. Many of them could not be connected normally. This is the connection LOG information of one of them.
[comlog.txt](https://github.com/arkypita/LaserGRBL/files/8396803/comlog.txt)

arkypita, 2022-04-01:
Could you share the sessionlog too?

[image]

CYCHEN97, 2022-04-01:
[sessionlog.txt](https://github.com/arkypita/LaserGRBL/files/8396849/sessionlog.txt)

arkypita, 2022-04-01:
here is your answer: 

```
01/04/2022 18:14:10.877	RealTimeStatus	Ex on [Idle|MPos:-30.000,-30.000,0.000|FS:0,0|Pn:XY] message
01/04/2022 18:14:10.877	RealTimeStatus	System.ArgumentException: 未找到请求的值“Idle|MPos”。
01/04/2022 18:14:10.877	RealTimeStatus	   在 System.Enum.TryParseEnum(Type enumType, String value, Boolean ignoreCase, EnumResult& parseResult)
01/04/2022 18:14:10.877	RealTimeStatus	   在 System.Enum.Parse(Type enumType, String value, Boolean ignoreCase)
01/04/2022 18:14:10.877	RealTimeStatus	   在 LaserGRBL.GrblCore.ParseMachineStatus(String data) 位置 E:\LaserGRBL\LaserGRBL\Core\GrblCore.cs:行号 2264
01/04/2022 18:14:10.877	RealTimeStatus	   在 LaserGRBL.GrblCore.ManageRealTimeStatus(String rline) 位置 E:\LaserGRBL\LaserGRBL\Core\GrblCore.cs:行号 1987
```

LaserGRBL is not able to parse (read and understand) the real time status message. I think that is because it is not able to correctly identify the grbl firmware version of your board and so it use the wrong logic to split and parse the realtime status.

The real time status message has changed in the history of grbl firmware versions, and so LaserGRBL should try to detect the version and implement different parsing algorithm.

The line of code GrblCore.cs: Line 1987 is inside the flow that handle the old version of grbl protocol (v1.0 and older) but the line it receive is not in the old format but in the new one, and so it is why it crashes.

I need to understand why it is not able to correctly detect the version.

The code that detect the version is:

```
private GrblVersionInfo StatusReportVersion(string rline)
{
	//if version is known -> return version
	if (GrblVersion != null)
		return GrblVersion;

	//else guess from rline
	//the check of Pin: is due to compatibility with 1.0c https://github.com/arkypita/LaserGRBL/issues/317
	if (rline.Contains("|") && !rline.Contains("Pin:"))
		return new GrblVersionInfo(1, 1);
	else if (rline.Contains("|") && rline.Contains("Pin:"))
		return new GrblVersionInfo(1, 0, 'c');
	else
		return new GrblVersionInfo(0, 9);
}
```

arkypita, 2022-04-01:
> My board uses TinyUSB of ESP32S2, without CH340 serial chip, to connect LaserGRBL

What board is it?

arkypita, 2022-04-01:
I found the issues, one issue in your machine, one issue in LaserGRBL.

When the machine connect to lasergrbl it should send the "Grbl welcome message" with firmware version, this is the best way for LaserGRBL to know the firmware version running on the board.

Your machine is not sending this information, this is the first issue.

In absence of this info LaserGRBL should try to guess the version from status message format, but it use v1.1 and then it test by "greater then" instead of "greater of equal". This is the issue in LaserGRBL.

I am going to build a fixup. Stay tuned.

arkypita, 2022-04-01:
Download and try this version: https://github.com/arkypita/LaserGRBL/releases/download/v4.8.1/install.exe

closed 2022-04-01 by arkypita

CYCHEN97, 2022-04-02:
> Download and try this version: https://github.com/arkypita/LaserGRBL/releases/download/v4.8.1/install.exe

@arkypita 
With this version, the problem still exists.
[comlog2.txt](https://github.com/arkypita/LaserGRBL/files/8401918/comlog2.txt)
[sessionlog2.txt](https://github.com/arkypita/LaserGRBL/files/8401919/sessionlog2.txt)

arkypita, 2022-04-02:
Sorry to tell you that I fix the LaserGrbl issue, but issues still exist in your controller.

As you can see by last comlog you upload, your controller is not sending any status report.

If you read what I wrote some post ago, without receiving data from the machine LaserGrbl should not be able to validate the connection. How LaserGrbl can send any kind of commands to the machine if the state of the machine is unknown?

Some post ago I suggest you to enable hard/soft reset on connection, but from log it looks like you didn't follow my hint. 

Is it clear to me that issues are in your controller.

arkypita, 2022-04-02:
What board is it? What do you know about the firmware it use?

CYCHEN97, 2022-04-02:
@arkypita 
I've tried resetting both hardware and software, but nothing works.The board is self-designed, GRBL version is like 1.3a.

arkypita, 2022-04-02:
So you are probably using a grbl from here? https://github.com/bdring/Grbl_Esp32
Look like that this firmware support different way of usage like WiFi, web interface, and more. Are you sure that it is not trying to communicate in a different way so it didn't reply over USB serial?

However I think that this is not a LaserGRBL issue, so I suggest you to google or ask here https://github.com/bdring/Grbl_Esp32 for more specific troubleshooting.

CYCHEN97, 2022-04-06:
@arkypita

arkypita, 2022-04-06:
busy with my job, please give me time

arkypita, 2022-04-06:
> I use QSerialPort of Qt host software written by myself to normally open for communication control every time, and LightBurn can also be used for communication control every time. 

In Windows OS only one software can open certain COM port, so be sure that other software doesn't use the engraver serial port when connecting to LaserGRBL.

> Is there any difference between LaserGRBL and them? 

I don't know LightBurn too much, but from what I know LightBurn doesn't do so much test as LaserGRBL on data sent from the machine. I mean: LightBurn user interface does not react to the states sent from the machine, as well as it does not show the current cursor position. As far as I know, lightburn could do some kind of one-way communication, just sending the gcode and ignoring what the machine says.

LaserGRBL instead decodes and interprets the status report sent by the machine, as I have already explained to you, to provide an UI that reacts to states. For example, the Jog buttons and the start-job button are automatically disabled when the machine is in alarm state.

LaserGRBL is written to be fully compatible with the original grbl firmware, according to the specifications and documentation you can find here: https://github.com/gnea/grbl/wiki/Grbl-v1.1-Interface

In order to be considered validly connected to a GRBL compatible machine LaserGRBL expects to receive:
- a [welcome message](https://github.com/gnea/grbl/wiki/Grbl-v1.1-Interface#welcome-message) (optional) but help to know what feature are available.
- at least one [status report message](https://github.com/gnea/grbl/wiki/Grbl-v1.1-Interface#real-time-status-reports) (mandatory) to know machine status. 

LaserGRBL actively ping the machine querying for status report message (sending the "?" character 0x3F) but if the machine did not reply with a status report message, LaserGRBL drop the connection after 10s considering itself not connected to a valid GRBL machine.

```
00000	00012033	log	Recording session started @ 2022/4/1 14:58:13
00001	00013670	com	Close COM8 [CORE]
00002	00017819	com	Open COM8 @ 115200 baud  (UsbSerial)
00003	00017828	tx	[3F]
00004	00019837	tx	[3F]
00005	00021837	tx	[3F]
00006	00023842	tx	[3F]
00007	00025829	tx	[3F]
00008	00027832	com	Close COM8 [CORE]
```

I don't know how else I can make you understand: your machine is NOT behaving like a GRBL device, so the problem is not in LaserGRBL but it is in the firmware running on yours TinyUSB of ESP32S2.

> Is the LaserGRBL communication only for the CH340?

No, LaserGRBL doesn't even know which serial chip he's talking to, he just needs a COM port.

commit 6df85a67 (2022-04-01): fix #1799