arkypita/LaserGRBL #1800: driver wont install

JeffG39, 2022-04-01:
LaserGrbl v3.8.0.0
Grbl v
Wrapper: UsbSerial
Win 10, Home, 10.0.19042.0 (2)
CLR: C:\Windows\Microsoft.NET\Framework\v2.0.50727\mscorlib.dll 2.0.50727.9151 (WinRelRS6.050727-9100)

TypeOf exception  [System.ComponentModel.Win32Exception]
Exception message [The system cannot find the file specified]
Exception source  [System], thread [Main Thread]
Exception method  [Boolean StartWithShellExecuteEx(System.Diagnostics.ProcessStartInfo)]

   ----------- stack trace -----------
   at System.Diagnostics.Process.StartWithShellExecuteEx(ProcessStartInfo startInfo)
   at System.Diagnostics.Process.Start()
   at System.Diagnostics.Process.Start(ProcessStartInfo startInfo)
   at System.Diagnostics.Process.Start(String fileName)
   at LaserGRBL.MainForm.installCH340DriverToolStripMenuItem_Click(Object sender, EventArgs e)
   at System.Windows.Forms.ToolStripItem.RaiseEvent(Object key, EventArgs e)
   at System.Windows.Forms.ToolStripMenuItem.OnClick(EventArgs e)
   at System.Windows.Forms.ToolStripItem.HandleClick(EventArgs e)
   at System.Windows.Forms.ToolStripItem.HandleMouseUp(MouseEventArgs e)
   at System.Windows.Forms.ToolStripItem.FireEventInteractive(EventArgs e, ToolStripItemEventType met)
   at System.Windows.Forms.ToolStripItem.FireEvent(EventArgs e, ToolStripItemEventType met)
   at System.Windows.Forms.ToolStrip.OnMouseUp(MouseEventArgs mea)
   at System.Windows.Forms.ToolStripDropDown.OnMouseUp(MouseEventArgs mea)
   at System.Windows.Forms.Control.WmMouseUp(Message& m, MouseButtons button, Int32 clicks)
   at System.Windows.Forms.Control.WndProc(Message& m)
   at System.Windows.Forms.ScrollableControl.WndProc(Message& m)
   at System.Windows.Forms.ToolStrip.WndProc(Message& m)
   at System.Windows.Forms.ToolStripDropDown.WndProc(Message& m)
   at System.Windows.Forms.Control.ControlNativeWindow.OnMessage(Message& m)
   at System.Windows.Forms.Control.ControlNativeWindow.WndProc(Message& m)
   at System.Windows.Forms.NativeWindow.Callback(IntPtr hWnd, Int32 msg, IntPtr wparam, IntPtr lparam)

StuartB4, 2022-04-02:
I would upgrade to the latest version of LaserGRBL and update Microsoft.NET Framework.

https://lasergrbl.com/download

https://dotnet.microsoft.com/en-us/download/dotnet-framework/net48

This is the one you need. 
[image]

StuartB4, 2022-04-02:
Did you have any luck with it?