arkypita/LaserGRBL #1806: preview window not centered on work image

Ams-Laser, 2022-04-04:
**Describe the bug**
A clear and concise description of what the bug is.
opening a gcode file that has x0y0 on center of work, gets cropped in preview window

**To Reproduce**
Steps to reproduce the behavior:
open attached file

**Expected behavior**
work preview is centered on preview window

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

**Hardware and software configuration (please complete the following information):**
 - Windows version
 -  10
 - Engraver brand and model
 - self build
 - Grbl version (i.e. 0.9j, 1.1f)
 - 1.1l
 - LaserGRBL version
 - 4.8.0

**Additional context**
Add any other context about the problem here.

arkypita, 2022-04-06:
According to GRBL definition (LaserON = M3/M4 and LaserOFF = M5) the gcode you attach never turn the laser on, so lasergrbl compute a job size of 0.0 x 0.0 mm that's why it is not centered in workspace

Ams-Laser, 2022-04-07:
isee, this is an embroidery G-code no need for laser or splindle since z is doing the stitches, and x y the movement, is it posible to center the design ?
or any code i might need to inject so the preview becomes right ?
i would love to use laser grbl to stream this type of code to my machine, i already use it for laser, plasma, oxifuel machines
thanks
Edward

arkypita, 2022-04-19:
> or any code i might need to inject so the preview becomes right ?

M3 is the code you can use to tell LaserGRBL the laser is on.

closed 2022-08-16 by arkypita