arkypita/LaserGRBL #1807: "Error Reading Config!"

mattymcnuggets, 2022-04-06:
Our setup was working perfectly last night, then all of a sudden we kept getting a "travel exceeded" error. We are running two stepper motors, a NEMA 23 on the x-axis and a NEMA 17 for the y-axis. The goal is fast travel in the x-axis mainly. The error happened while increasing the mm/min of the X-axis when trying to max out its speed. We ran it until we figured the X-axis was perfectly configured (didn't actually max out to stalling, max speed at about 20000 mm/min and 500 mm/m^2 acceleration, feed rate of 7500) then we changed some settings in the Y-axis because we thought we might as well try to increase the speed of that axis too. 

After increasing speed and acceleration in the y-axis (and also increasing feed rate in the g-code to 10000) we started getting the error. We've unhooked all of the limit switches and now we don't get the original ''travel exceeded error, but the x-axis no longer responds to GRBL commands. We can jog the y-axis which is good, but there's nothing from the x-axis. The x-axis motor barely feels warm to the touch, a far cry from how hot it usually is even when idling (is requires 36 V). 

Additionally, when trying to change the config in GRBL, we get an error "error reading config!" and are unable to make config edits and write them as usual. This is a very puzzling issue in that we had the etcher working perfectly right before getting this error. From inspection, all our hardware looks to be in working condition still (lights show up on all drivers and the arduino) and it doesn't look like there's any wires that were accidentally unplugged but we certainly could've overlooked some things.

arkypita, 2022-04-06:
Hi, maybe your macine has a damaged firmware or a damaged configuration.

LaserGRBL usually reply with error "error reading config!" when the machine didn't reply with the numbers of expected config parameters, or give some unreadable reply in a given time.

If you cannot read the configuration from the "Grbl Configuration" form you can try to access to configuration via $$ command i "type gcode here" box.

[image]

You can also set $ values from this box by typing "$CODE=value"
Is it possible to send the machine a reset config by giving "$RST=*" command (but please write down your config somewhere before resetting).

Let me know

mattymcnuggets, 2022-04-06:
Hey the $RST=* command worked and everythings running normal again. Thank you for the quick reply!

closed 2022-04-06 by mattymcnuggets